prusa3d/PrusaSlicer #756: [Request] Add optional perimeter to raft

SamuelSVD, 2018-03-05:
### Version
1.38.6-prusa3d-win64

### Operating system type + version
Win 10 Pro

### Behavior
* _Describe the problem_:
When using a raft, even when configured correctly, the quick back and forth motion of the Prusa i3 Mk2 when it reaches the limits causes the raft to lift.
*_Current problem_:
Here is where the raft lifts:
[image]
* _Expected Result of Request_:
A perimeter around the raft should stop the lifting of the raft when it reaches the ends as the printed material should adhere to the perimeter:
[image]
This could be implemented as an optional  parameter (a checkbox) that only works when raft layers is greater than 1.

_Is this a new feature request?_
Yes
#### STL/Config (.ZIP) where problem occurs
[Request.zip](https://github.com/prusa3d/Slic3r/files/1779351/Request.zip)

bubnikv, 2018-03-15:
Discussed with Jindrich, our print expert and Slic3r profile tweaker, he thinks we should make it a default feature without any additional checkbox.

Should we print the perimeter before or after the usual 1st layer support?

SamuelSVD, 2018-03-15:
I think we should print the perimeter before the usual 1st layer support. This way the first layer will be able to adhere to the perimeter, reducing the probability for error.

bubnikv, 2018-03-15:
> I think we should print the perimeter before the usual 1st layer support. This way the first layer will be able to adhere to the perimeter, reducing the probability for error.

Jindrich has the same opinion.

bubnikv, 2021-02-24:
Implemented with fcb714cd241f0847522b9aac57da6b62acdd0152

The 1st layer support or raft will always have a perimeter drawn around it.

Also there will be new parameters raft_first_layer_density and raft_first_layer_expansion
to influence the 1st layer of raft or support.

closed 2021-02-24 by bubnikv