prusa3d/PrusaSlicer #757: Incomprehensible wording of help tooltip for "external perimeters" field.

BETLOG, 2018-03-05:
### Version
Slic3r-1.39.1-beta-prusa3d-linux64-full-201802131106

### Operating system type + version
Linux betlog 3.13.0-142-generic #191-Ubuntu SMP Fri Feb 2 12:13:35 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
* _Describe the problem_
Incomprehensible wording of help tooltip for "external perimeters" field.
[image]

"...If left zero, default extrusion width will be used if set, otherwise 1.125% nozzle diameter will be used"
So if i put 0 in the field do I get default nozzle width..... or 1.125% nozzle width?

tkircher, 2018-03-06:
Let's break it down. "Set this to a non-zero value to set a manual extrusion width for external perimeters." So, if it's zero, the extrusion width for external perimeters is calculated automatically rather than set manually. "If left zero, default extrusion width will be used if set, otherwise 1.125x nozzle diameter will be used." So, this is talking about the automatic calculation, because if you set the value manually, it will use that value. Consequently, if the default extrusion width is set, it uses that. But if there's no default extrusion width set, then it will calculate the external perimeter value by evaluating 1.125x the nozzle diameter. So, what happens not only depends on the value in this field, but the value in the default extrusion width field. Hope that clears things up.

BETLOG, 2018-03-06:
> But if there's no default extrusion width set, then it will calculate the external perimeter value by evaluating 1.125x the nozzle diameter. So, what happens not only depends on the value in this field, but the value in the default extrusion width field.
Perfect.
I'd like to  suggest that similar wording is added to the help tooltip.

BETLOG, 2018-03-06:
Hopefully  clearer wording.

>Set  this to a non-zero value to manually set the extrusion width for external  perimeters. Or set this to zero and default extrusion width will be used, however if default extrusion width is not manually specified then 1.125x of nozzle diameter value will be used.

rtyr, 2021-01-20:
I think it is OK as it is. Closing.

closed 2021-01-20 by rtyr