prusa3d/PrusaSlicer #758: brim for the skirt

adlerm1, 2018-03-05:
Good Morning!
It would not be bad to add the ability to create an brim for the skirt.
It is very important for printing plastics type ABS without a full-fledged thermal chamber. Because without the brim, the skirt often tears off the table.

gregspruce, 2018-03-05:
You could increase the number of loops/min distance on the skirt. You could omit the skirt altogether and put a brim on the part.

adlerm1, 2018-03-05:
When printing with ABS plastic, so that the part does not crack, it is very convenient to lift the skirt in height close to the height of the part, but the skirt does not often tear off the table, which leads to problems.
varying the settings it is not possible to combine the brim and the skirt.
If there was an opportunity to create a brim for a skirt, it would be excellent.
[image]
[image]

lukasmatena, 2018-05-22:
A duplicate (or almost) of https://github.com/prusa3d/Slic3r/issues/724. See the discussion there for recent updates.

I'm closing this issue, if that's ok.

closed 2018-05-22 by lukasmatena