prusa3d/PrusaSlicer #759: how to make a temp tower? (feature to add may be ?)

blandiant, 2018-03-05:
### slic3r Version 1.39.1

### Operating system type + version
 windows 10
### Behavior
Hi
I want to make a temp tower in slic3r but i can't find any parameter to do that, i tried this script (https://www.thingiverse.com/thing:2615842) but i can't get it to work, is there a solution directly in slic3r without having to edit the gcode?  
if not, it would be nice to add this feature : a parameter (or an external software may be) to change the temperature at every X layers (or mm) from X degrees each time or to have more control: to choose oneself the layers from which we change the temperature and  the new temperature to print with.
thank you

CapnBry, 2018-03-06:
This is easily done with a gcode snippet in the "Before layer change" or "After layer change" script:
```
{if layer_z > 30}M104 S240
{elsif layer_z >20}M104 S230
{elsif layer_z > 10}M104 S225
{endif}
```

Just change the layer_z to where you want to make each transition and the S number to what temperature you want, starting from the top down. I just tried this in the latest pre-release and it works great.

blandiant, 2018-03-06:
hi, i just tried it, but it doesn't work, so i tried to change the temperature myself during the print (in the "tune" section) but after few second (each time) the 3d printer go automatically back to the temperature set in slic3r...

CapnBry, 2018-03-06:
I'd check the gcode before you say "It doesn't work". Also, Slic3r only sets the temperature twice in the whole gcode file normally, once on the first layer and once when transitioning to the second layer so there's no way it is changing the temps on you (unless your conditional gcode is telling it to).

blandiant, 2018-03-06:
i checked the gcode, it's writen in the gcode : before_layer_gcode = ;BEFORE_LAYER_CHANGE\n;[layer_z]\n{if layer_z > 10}M104 S265\n{elsif layer_z >17}M104 S260\n{elsif layer_z > 24}M104 S255\n{elsif layer_z > 31}M104 S250\n{elsif layer_z > 38}M104 S245\n{elsif layer_z > 45}M104 S240\n{endif}
but the temperature doesn't change during the print 
and when i tried to change the temperature manually,but  after few the printer go back to the 2nd temperature: for the first layer i set 270 and for the others layers i set 260, when i changed the nozzle temp manually to 250 in the "tune" menu, the printer go back automatically to 260 after few second

CapnBry, 2018-03-06:
Check the GCode for M104 lines and see if the conditional gcode is coming out. I've also noticed you've got your gcode backwards. It needs to be from layer_z highest to lowest, not lowest to highest like you've got it (because the first if statement is true for all layers > Z=10 so none of the other conditions will fire). If you want to go from low to high you've got to flip the conditional.
```
;BEFORE_LAYER_CHANGE
;[layer_z]
{if layer_z < 10}M104 S265
{elsif layer_z <17}M104 S260
{elsif layer_z < 24}M104 S255
{elsif layer_z < 31}M104 S250
{elsif layer_z < 38}M104 S245
{elsif layer_z < 45}M104 S240
{endif}
```

blandiant, 2018-03-06:
i checked and like and all the m104 set the temp to 265 because i set the conditional in the wrong direction
and this is why when i changed manually the temp, the printer goes automatically back to 265 
Thank you a lot

bubnikv, 2018-03-07:
You can also write an expression in a form like

M104 S{if layer_z < 10}265{elsif layer_z > 45}240{else}265+(240-265)*(layer_z-10.0)/(45-10){endif}

I did not test it myself though. The idea is to interpolate the values between layers 10 and 45 linearly. Please note, that the number 10.0 inside (layer_z-10.0) is written as a real number, which then forces the whole expression to be evaluated in real numbers. Without that, the whole expression would be calculated in integers, leading to rounding in the division. So this expression will print a real number with decimals into the G-code.

One can also use the ternary operator (again, I did not test it, you have to test it yourself), and in the future, we may introduce the min / max functions, and maybe a linear interpolation function would be quite useful as well.

M104 S{(layer_z < 10) ? 265 : (layer_z > 45) ? 240 : 265+(240-265)*(layer_z-10.0)/(45-10)}

bubnikv, 2018-03-14:
Solved by the conditional G-code. Closing.

closed 2018-03-14 by bubnikv