prusa3d/PrusaSlicer #760: Preview:  Click "Show: Feature types" and "Perimeter" gets unchecked the first time (even though they were visible)

VanessaE, 2018-03-05:
### Version

commit 3c64eb9215cc6d285571866062d933035d435eb6

### Operating system type + version

Debian 9.3

### Behavior

Title says it all:  
[image]

#### STL/Config (.ZIP) where problem occurs
Any random model, config should be unimportant.

bubnikv, 2018-03-15:
Verified by @FidelCapo with the integration build 1.40.0-alpha.249.
@enricoturri1966 Would you please look into this? Thanks.

lukasmatena, 2018-05-17:
Should be fixed by https://github.com/prusa3d/Slic3r/commit/6f792b7ffbc90ef535ba7ced09cf77a8110e87c0.

Closing the issue, reopen it if problem persists. Thanks for the feedback.

closed 2018-05-17 by lukasmatena