prusa3d/PrusaSlicer #761: Inside-out confusion?

Mike4U, 2018-03-05:
### Version
_1.39.1-beta-prusa3d
OS 10.13.3 

### Behavior
*Outer perimeters often print faster than inner.
[image]
* _Steps needed to reproduce the problem_
  Seems to happen on slightly complicated parts
* _Expected Results_
 Slower outside perimeter 
* _Actual Results_
  *Faster outside, sometimes changes on different parts of part
[gear.zip](https://github.com/prusa3d/Slic3r/files/1782397/gear.zip)
Thanks

Sebastianv650, 2018-03-05:
It's part of the cooling logic. External perimeters will be printed add long as possible with normal external perimeter speed, everything else will be slowed down for cooling.

Mike4U, 2018-03-05:
Thanks,
I'll have to look at some of my files, perhaps layer size was the reason.

closed 2018-03-05 by Mike4U