prusa3d/PrusaSlicer #762: Cutting a part should NOT re-arrange the other parts on the plate

VanessaE, 2018-03-07:
### Version

master branch, commit 4a90ab1f6a1390df4ce332b342536ee61db6ebfd

### Operating system type + version

Debian 9.3

### Behavior

When you cut a part, Slic3r re-arranges ALL of the parts on the plate, whether they're affected by the cut part or not.  This screws up situations requiring precisely-aligned parts (think multi-color or custom supports).

If a cut operation keeps only the upper or only the lower portion of the cut piece, the kept part should simply take the place of the original.  If both parts are kept, Slic3r should just put the lower part where the original was, and look for a suitable place to put the upper part.  If it can't find a good spot, just put it off-plate, and let the user re-arrange things manually.

In no case should any part on the plate be moved.

#### STL/Config (.ZIP) where problem occurs
Any two or more random models, no config necessary

[ This issue has been copied over from upstream alexrj/Slic3r as it affects this fork also ]

morphias2004, 2024-04-25:
+1

Still there in PS 2.7.4

morphias2004, 2026-09-18:
Is this fixed in PS3, because it's still there in PS 2.9.6?