prusa3d/PrusaSlicer #764: Always open the "Add file" dialog on the same screen as the main window

VanessaE, 2018-03-07:
### Version

master branch, commit 4a90ab1f6a1390df4ce332b342536ee61db6ebfd

### Operating system type + version

Debian 9.3, with XFCE 4.12 (in case the desktop environment matters)

### Behavior

For me, no matter what I do, Slic3r insists on putting its "Add file " dialog on screen 1 (of 3), regardless of which screen Slic3r itself is on, usually screen 3).  It even does this if screen 1 has a program running in full screen mode (e.g. watching a movie).

I guess Slic3r is passing some window manager hint to force this, but I have XFCE's window manager already set top open new windows wherever the cursor is.

For comparison, Pronterface'ds "Load" dialog behaves exactly as it should.

#### STL/Config (.ZIP) where problem occurs

None needed

[ This issue has been copied over from upstream alexrj/Slic3r as it affects this fork also ]

vojtechkral, 2018-05-14:
This doesn't happen on my Linux box. Slic3r is definitely not forcing this on purpose. Probably some kind of a mixup on the wxWidgets - GTK - windows manager axis...

bubnikv, 2018-05-16:
Do you have a similar issue with other applications on your XWindow system? Or is it just Slic3r. It looks to as like a general issue of your window manager.

VanessaE, 2018-05-16:
It does happen on one or two other programs, but in most cases, new windows are opened where the pointer is.  Those that don't, always put it waaaaaay over on the left screen (out of 4).  I assume it has to do with that desktop layout putting that screen at the desktop's lowest X coordinate, and windows opening over there either have, or lack, some important window manager hint.

VanessaE, 2018-05-16:
It only appears to affect the "Open STL/OBJ/AMF/3MF..." dialog (whether I call it up from the File menu or the "Add..." button, and the "Open 2.5D TIN mesh" dialog in the "File" menu.

All other dialogs open on the correct screen - print/printer/filament profiles, all four export options in the Plater menu, all of the File menu's items except the above, all of the relevant items in the right-click-on-the-part menu...

bubnikv, 2018-08-02:
Sorry, we were not able to reproduce your issue on none of our Linux systems around. This must be some quirk of your setup I suppose. Closing.

closed 2018-08-02 by bubnikv