prusa3d/PrusaSlicer #765: Auto-centering should not invoke auto-arrange

VanessaE, 2018-03-07:
### Version

master branch, commit 4a90ab1f6a1390df4ce332b342536ee61db6ebfd

### Operating system type + version

Debian 9.3

### Behavior

If you turn auto-centering on, it does not actually auto-center parts added to the plate.  Rather, as each part is added - whether you add one file at a time, or several at once - the plate is auto-arranged after loading each part, thus defeating the purpose of the auto-center feature.

With that option turned off, parts are added to the middle of the plate, centered according to their geometry.  So basically, this option behaves opposite of how it should.

What *should* happen is that parts are placed with their *model origin* snapped to the center of the plate, period, even though they'll overlap.  Auto-arrange should not be invoked unless the user explicitly clicks the "Arrange" button at the top of the window.

That would allow the user to place multiple files on the plate that are already aligned in their modeling program, such as a multi-color model or a model with its own custom support objects (assuming all objects in a given model have their origins set to the same coordinates).

Ideally, if you add two or more parts at the same time, Slic3r should place them at the middle of the plate, perhaps according to the geometry of the largest part of the set, but then the smaller part(s) should be snapped to the same relative position according to their respective model origins.  In the two parts in the zip below, the origin has been set (in the modeling program) to the center of the crosshairs (a sort of "registration mark") that is formed when the parts are precisely aligned.  Here, you can see the parts properly aligned (by hand after adding).  The light green dot near my pointer in the image is a tiny cylinder I added to the plate to mark the exact center, while you can see the crosshairs below and to the left of it inside a square void in the part:

[image]

#### STL/Config (.ZIP) where problem occurs

No config file is needed here.  Model files, for evaluation of how it should work (imho):   
[frame.zip](https://github.com/prusa3d/Slic3r/files/1789059/frame.zip)

[ This issue has been ported over, with significant changes, from upstream alexrj/Slic3r as it affects this fork also ]

bubnikv, 2018-08-02:
@tamasmeszaros Would you please investigate this request? Thanks.

tamasmeszaros, 2018-08-09:
Dear Vanessa,
currently, we know about the clash of auto-center and the arrangement feature. Auto centering is taking the parts according to their bounding boxes (although I'm not entirely sure about this). Arrangement is a completely different beast altogether but when you hit the arrange, it does center the arranged pile of objects within the print bed, that is why we kept this logic for now. 

The future of the auto-centering is under negotiation and as to whether it should work according to the model origin, we will consider your proposal as well.

Best regards,
Tamás

VanessaE, 2019-01-03:
Just FYI, I no longer use Slic3r PE, just upstream.   However, this is still an issue there.

> Would you please explain what should be the difference in your opinion between the Auto Center on / off?

I'm not sure how to explain it better, but I'll try.  Here is the behavior of Slic3r right now:

**If I have auto-centering turned off:**

Loading one or more parts (any number) places them every which way around the plate, often off-plate entirely, and just as often, at the front-left corner, where (0,0,0) is on my printer.  Where they end up on Slic3r's plate depends on where they were in the world/scene in whatever program the model was made in, even if each object's origin is carefully set to a meaningful spot near, say, the bottom-center of the object.  If I move an object to the right in Blender (that is, positive along the X axis), for example, and then export to STL and load it in Slic3r, it will be displaced to the right on the plate by the same amount.

Example:

[image]

Every one of these objects has its origin somewhere near its bottom-center.  Some were exported while near the world/scene center point (those ended up at the corner of the plate), others not so much (like the gear and other objects in the lower right).  As you can see, all but ONE part is either partially or entirely off-plate.

What you can't clearly see in this image is that the objects that need precise alignment *are in fact aligned* (because I was careful to always keep things aligned and rotated the same, and exported one after the other), whether on-plate or off-plate.

If I move one of the parts that was loaded, it's simply moved, and nothing weird happens.

**If I have auto-centering turned on:**

Loading one part moves it to the center of the plate, regardless of where it was in the world/scene.  This is precisely the correct and desired behavior.

However, loading multiple parts automatically rearranges them as each part is loaded, with Slic3r re-centering the group as a whole before loading the next selected part.  So if you select and load, say, 10 parts, auto-arrange and auto-center-the-whole-plate are invoked 10 separate times, before finally passing control back to the user.

Example:

[image]

Here, I cleared the plate, enabled auto-centering, and loaded up the same group of parts.

**What I find wrong:**

Why must Slic3r move the parts around at all?  Why can't Slic3r just load them all at the center of the plate, letting them overlap, and let *me* decide if things need to be moved around?  What word or phrase in "Auto center parts (x,y)" can possibly mean "auto-rearrange all parts on the plate each time a new part is added" ?  

Why, when auto-centering is disabled, is Slic3r using the origin of the plate as the reference when placing a part, instead of the plate's geometric center?  If this were a delta printer, the origin would be in the center of the plate, but in a cartesian, it's virtually a certainty that the place origin will be in one corner.  Blindly using the plate origin guarantees that most parts will end up partially off-plate when loaded with auto-center disabled.  Using the geometric center of the plate (regardless of shape or origin) would guarantee that parts almost never end up off-bed when loaded.

If I have one part and try to move it while auto-centering is enabled, Slic3r just snaps it back to center, so in essence, I can't move the part in this mode.  If there's more than one part on the plate, moving one of them causes Slic3r to re-center the plate's contents after I let go of the button, making it impossible to, let's say, load two parts and put one at the plate center, and the other off in a corner somewhere.

**What did I expect from auto-centering mode:**

I'd prefer parts to be dropped in the middle of the plate when loaded, and never moved from there unless I explicitly drag one with the mouse, or click "Arrange".

After loading a bunch of parts and disabling auto-centering so that I could move things around to make a big pile :smiley: :

[image]

This is what I would have expected auto-centering to do without my help.  Clearly no one would load all of those parts all at once, perhaps instead choosing to load a few parts, move them away from center, then load up the parts that need precise alignment.  Naturally I wouldn't leave the plate in this state were I to export to G-code, and yes, one of the parts goes off-plate (but that's my fault - I need to choose a better place to put the origin for that object and its partner), but it demonstrates what precise alignment of all loaded parts would look like, if positioned according to their individual origins. That's easy to do by hand in this horribly-contrived example, since I know where I put all of the models' origins.

After moving a couple of parts out of that mess that I intentionally created, everything is left undisturbed:

[image]

Now, if I turn auto-centering back on and move a couple of the existing parts:

[image]

That's not so bad, really, as the whole plate moved as a group, leaving the arrangement of parts otherwise undisturbed.

However, if I now load a new part onto the plate, with auto-center still turned on:

[image]

FAIL.

**My recommendation:**

"Auto center parts (x,y)" should should be split into two self-explanatory options:

"Actively keep plate contents centered (x,y)"
"Auto-rearrange plate after loading a part"

> you should load them as parts of a multi-part object

[... cut ...]
**Bottom line:**

All parts should be placed with their object origins at the plate's geometric center on-load, regardless of where the plate's (0,0,0) is, and I should be able to freely load more objects and move things around afterward without having Slic3r interfere with my actions or alter the parts' arrangement.

Right now, there is no combination of settings and actions that allows for this.

(apologies if this comes off as snotty, that's not my intent; I'm just trying to get as clear as I can)

VanessaE, 2019-01-03:
Sorry for being so long-winded :smiley:

VanessaE, 2019-02-11:
@bubnikv any further ideas?

(fwiw, this applies perhaps more to upstream as to the prusa3d fork, as I haven't used prusa3d in a while)

bubnikv, 2019-03-01:
@VanessaE We are working on a solution similar to what Cura does: Newly added objects should be auto-arranged around the center of the print bed without collisions.

For the scenarios where you want to arrange multiple volumes to touch or overlap, it is now better in Slic3r 1.42 to include these volumes into a single object.

By the way, we did a huge amount of UI rework in the current Slic3r PE master, we recommend you to test it.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha5

and alpha7 is due today.

VanessaE, 2019-11-21:
not a problem anymore

closed 2019-11-21 by VanessaE