prusa3d/PrusaSlicer #766: Feature request: "prime/dwell tower" as alternate/extension to auto-cooling "slow down"

VanessaE, 2018-03-07:
### Version

master branch, commit 4a90ab1f6a1390df4ce332b342536ee61db6ebfd

### Operating system type + version

Debian 9.3

## Behavior

Right now, Slic3r has the option to slow down the feed/flow rates of a layer to extend the layer time to meet a configured threshold.  That works for some things, but not always for parts with small horizontal cross-sections, because creeping along too slowly in a small area, bumping to the next layer and repeating over and over as the part grows just dumps too much heat into that section, overheating it, and defeating the purpose of the slow-down feature.  This problem is compounded when you use thinner layers.

In such cases, users are usually told to add more parts to the plate (or more copies of the single part), but that wastes filament if the user only needs the one part.  Or, users are directed to add "sacrificial" columns/towers, but these require adding several custom settings, and they waste time if only a few layers need the extra delay.

To my mind, the best solution would be to have Slic3r generate a custom tower, to which several features would be applied when the user slices the model. It should be a cylinder, around 20 mm in diameter, with a sizable number of faces, to avoid having any sharp corners that could cause it to lift off of the plate, and each of its layers would vary in print time as needed.

* The tower would be drawn with two bottom solid layers, no top solid layers, at least one perimeter, and possibly sparse, "aligned rectilinear" infill at 10-15% density.
* At the start of each layer, Slic3r would add a command to move the nozzle to the exact center of the tower, before drawing anything on the layer.
* Slic3r should then prime a tiny amount (a few tenths of a millimeter), then proceed to print the perimeters slow enough to create the necessary delay, then do enough normal-speed sparse infill to re-balance the nozzle pressure to be equal to the first lines of the actual part.
* Finally, a quick retract-travel-unretract to move to and start drawing the actual part island(s).

This would entail a couple of small changes to the Filament profile "Cooling" page:

* "Slow down if layer print time is below..."  would be re-worded to something like "Extend layer time to at least...".
* A single checkbox would of course be added to enable the tower.  When disabled, Slic3r would use its usual slow-down code, as needed, with no changes from the current behavior.  When enabled, Slic3r would still use its slow-down code on the actual part(s), and would use the tower to further extend the layer time to the configured minimum, if the user's configured "Min Print Speed" would still be too fast.  My reasoning here is that extremely small cross-sections are probably the result of a tall, thin section such as a clip-on mounting stud, and such things need to print quite slow to help keep the heated filament from pulling sideways on the part too hard, bending it around as the nozzle moves around.  If "Min Print Speed" is set to zero, disable the slow-down entirely and use only the tower to extend the layer time.  Default should be disabled.
* Maybe a field to configure the number of perimeters for the tower.  Default 2.  Or just hard-code it.

The tower should automatically appear on the plate as soon as the checkbox is turned on, to allow the user to move it as needed before slicing.  It should be displayed in some unique color.  Right-clicking the tower in the "3D" view should do nothing, or perhaps pop up a dialog to let the user adjust the minimum layer time.  A left-click should display something in the "Info" section on the right to indicate that it is a prime tower, as if the unique color weren't enough of an indication. :stuck_out_tongue: 

If the user slices the model, and it turns out that no part of the model actually needs a delay, then the tower should be completely omitted from the sliced results and its color in "3D" view should change to grey or something, at least until the user changes a relevant setting.  An appropriate notice should be added to the "Print Summary".  As an aside, it might not be a bad idea to add a notice for when slow-down gets invoked, as well (if the tower isn't being used).

Theoretically, one could set the skirt to go the full height of the object, but aside from probably using more filament than a tower, this would trap heat near the model, defeating the purpose (at least if you're printing with PLA or PETG).

Also note that if you're using especially wide traces, you may run into excess pressure at retract, resulting in more stringing/ooze than you'd otherwise expect (but stuff like Marlin's Linear Advance should be able to address that).

I understand that this fork has some kind of wipe/prime tower which I guess is used for multi-extruder setups, but aside from it not obviously being available for single-extruder bots, I guess it doesn't work like the above proposal.

## Stl/model

N/A

[ This issue has been copied over, with minor changes, from upstream alexrj/Slic3r as it affects this fork also ]

bubnikv, 2018-05-10:
We have had a discussion today about our wipe tower adjustments for the dual extruders. This topic is similar, maybe @lukasmatena could look into it.

wavexx, 2018-06-26:
I'd like to chime in on this. I have similar issues when printing small features, such as a single cylinder going upwards. Cura has a "Lift Head" option that we should copy.

The "slow down" ruined the print in any case it kicked in. The overheating caused by the slow-down remelts the top surface of the thin support, causing blobs and gaps.

Is there any case where the slow down is beneficial? In my tests, it never was.

Cura has a "Lift Head" option in addition to minimal layer printing time. Similarly to Slic3r, Cura tries to slow down if the layer printing time is short. However, with the "Lift Head" setting, the extruder is just lifted and parked until the minimal layer time is reached.

In my experience, this works *way* better.

It's still not optimal though: if there is too much waiting, depending on the material, the nozzle will ooze a bit, causing other kind of defects (under-excluded starting points and extra deposited material). The print quality of the piece though was always way better than slowing down.

For perfect results I also add a sacrifical tower somewhere in the print bed.

closed 2026-09-29 by prusaslicer-bot[bot]