prusa3d/PrusaSlicer #767: Overhang handling suggestions

VanessaE, 2018-03-07:
### Version

master branch, commit 4a90ab1f6a1390df4ce332b342536ee61db6ebfd

### Operating system type + version

Debian 9.3

### Behavior

Slic3r just does not handle overhangs very well... this *really* needs fixed.  Overhangs have been somewhat of a thorn in the side of Slic3r users for quite some time.  Related  issues:  alexrj/Slic3r#328, alexrj/Slic3r#2158, and alexrj/Slic3r#2313, and #242 in this fork.

After doing a *lot* of testing, I've made a list of overhang settings and behaviors that I think should be added. I know you don't like adding new settings, but in this case I think it's unavoidable.  These should all be applied to (and only to) partial overhangs, not bridges:

**[Print Settings]**
* Overhang print speed - on overhangs, like many things, slower is better, to a point, as it gives the filament more time to cool into the right shape.  Go slow enough and you can draw nearly-horizontal lines in mid-air (if you have enough cooling).  Obviously one could just slow down the whole print, but that wastes time.  Suggested default:  20 mm/sec.
* Overhang acceleration - a given printer might produce nice, sharp, non-overhanging corners even with very high acceleration most of the time, but high acceleration around an unsupported corner will surely cause the extruded filament to pull itself inward, especially if it has a large die swell.  Like speed, one could set a low acceleration for the whole print, but that wastes time below a certain point.  Suggested default:  100 mm/sec²
* Overhang line width - For overhanging perimeter lines that still have at least some overlap with the previous layer, wider lines tend to produce better surface quality on the underside of the overhang, but one surely does not want to use extra-wide perimeters around the rest of the part. Suggested default: automatic, nozzle diameter times 1.75.
* Overhang exterior first threshold - For any island that has overhanging perimeters, if the total length of the longest overhanging perimeter is less than this value, that island will have its exterior perimeter printed first.   Any islands whose perimeters exceed this value would have their exterior perimeters printed first or last according to the global checkbox.  My reasoning here is that small overhangs tend to work better with exterior perimeters printed first, but long overhangs (such as a beveled edge on the bottom of a project box) tend to warp less if the exteriors are printed last.  Suggested default:  50 mm.
* A checkbox to force concentric infill for the unsupported parts of the overhang's bottom solid layers (with the user's normal solid infill pattern just inside of that).  My theory here is two-fold:  1) long, narrow rectilinear infill concentrates too much heat at one end of the area, with the hot zone moving along as the infill is drawn, causing the overhang to want to curl sideways, while concentric spreads the heat fairly evenly, and 2) overhanging perimeters *will* droop some no matter what you do, so for a particularly steep overhang, the perimeters that rectilinear infill should anchor to just won't be where Slic3r expects, while concentric infill is already anchored at each "end" where the loops go in/out of the overhang.  Global, super-thick walls have a similar effect, but that wastes filament.  Suggested default:  enabled.

**[Filament Settings]**
* Overhang flow ratio - for overhanging lines that have no overlap with the previous layer (i.e. because they're too close to horizontal), less flow may be better, at least for some filaments (think ABS versus PLA).  There will likely need to be a setting for these that is different from bridging flow ratio.  As an aside, bridging flow ratio should be moved to filament settings, for the same reasons.  Less flow of course would cool faster, but this would depend on the filament type, temperature, and cooling settings.  Suggested default:  1.0 (what is is now).
* A threshold setting to decide how much of an overhang is needed (as a percent or degrees, like the support overhang threshold), before invoking the above flow ratio.  Suggested default, 60 degrees (basically what it is now).
* Overhang fan speed - some filaments can't have the fan at 100% all the time (again ABS versus PLA).  Suggested default:  100%
* Minimum fan spin-up time (or distance) - since fans need time to spin-up, and some filaments or print jobs can't have them on at 100% all the time, they need to be turned on or up some distance before the printer reaches the overhang, and turned off/down again shortly thereafter.  Since this will ultimately boil down to some amount of printed lines before and after the overhang, it might be useful to consider the ends of this range as the points where Slic3r would apply the various other suggestions above, if its seam-placement algorithm can't find a suitable place within some reasonable distance of this spot.  Suggested default:  1 second or 10 mm.

These two test models (especially the first one) are very good for evaluating these sorts of settings:
http://www.thingiverse.com/thing:1564848
http://www.thingiverse.com/thing:58218

By using enough global settings to simulate some of the above overhang-specific settings, good overhang quality can be achieved, at least with 0.2 mm layers.  Here's an example of what I was able to produce (the settings, below, are somewhat conservative compared to what I might use in practice, should these suggestions be implemented):
[image]

Example config (this is what was used to produce the parts in the photo above - do not use for a real print, it'll waste filament and/or time):  [segfault-config.ini.zip](https://github.com/alexrj/Slic3r/files/1002292/segfault-config.ini.zip)

Another point of note, which doesn't need a setting but which is a ...ahem... corner case :smile:  

Aside from never putting a seam on an overhang, don't put one close to the overhang either (the above "criss-cross" override notwithstanding).  For example, on the "XYZ" test cube pictured above ([XYZ_20mm_calibration_cube.stl.zip](https://github.com/alexrj/Slic3r/files/1005404/XYZ_20mm_calibration_cube.stl.zip)), if you use the "Aligned" seam position, Slic3r moves the seam around, depending the features on the sides of the cube.  For those layers that include the lower half of the "X" on the cube front, Slic3r consistently puts the seam on the lower right leg, inside right corner:

[image]

Red marks indicate the moves as they exist now, with the travel ending at the red "X" and printing proceeding forward and then to the right, as per the bent arrow, leading to a sharp, distorted edge at the overhanging corner that follows.  Blue marks indicate the path I would have expected Slic3r to use.  Of course, moving the seam as suggested there would move the distortion to the inset, but then it only affects a flat vertical surface, not an overhang.  One's filament choice has an effect here, likely due to viscosity and related to print temperature (the white that I use is much trickier to get "just right", compared to the other colors).

Of note, after doing a whole lot of tests (thankfully without a whole lot of wasted filament :stuck_out_tongue: ), one thing I've just about settled on is that as layers get thinner, the maximum overhang you can draw gets less and less.  At 0.2 mm layers, I can do a full 30-85 degree overhang arch with only moderate curling, but at 0.1 mm layers, it loses corner quality beyond 45 degrees (as expected), and starts to curl up too much by about 55 degrees (i.e. the hotend knocks the part off the plate).

[ This issue has been copied over from upstream alexrj/Slic3r as it affects this fork also ]

Sebastianv650, 2018-03-07:
I'm looking forward to see something like this implemented.

And I'm quite happy that LA seems to be somewath "VanessaE approved" that you now switched over to Slic3r to point out some issues here instead of Marlin :blush:
It's always helpful to read your reports :+1:

VanessaE, 2018-03-07:
haha wiseguy :stuck_out_tongue_winking_eye: 

I'm glad my ramblings are useful :smile:

rennne, 2019-02-02:
Another overhang improvement:

- avoid seam (retraction with z-lift) near overhang

I observed while extruding on some more extrem overhangs, that ending of the movement on an overhangs, tends to pull the corner where the z-lift and retraction happens (curling up). 

Then for next layer the overhang gets quite ugly because it has to extrude on these hardened bump on the corner, and cannot attach cleanly.

I comparison when the nozzle start and ends on the other side (there is no overhang on my model) it looks nearly perfect. 

I had a different problem with that: because of the hardened bumps on the overhand my print got kicked over. So this then would also be improved.

gnydick, 2019-11-11:
How about we break this up into a ticket per feature so they can get done faster. I think proportionally slowing down based on overhang angle would be a great low hanging fruit compared to the other suggestions.

anerb, 2019-11-20:
Hi, I'm a bit late to this thread.  Can I please add this suggestion/request:
In case of overhangs, print the outer perimeter last.
If the inner perimeter/infill is not overhanging, the outer perimeter will have something to hold onto (from the side) for a better chance of remaining at the correct z location as it is being extruded. Thanks.

gnydick, 2019-11-20:
Outer perimeters always print last unless you change that from its default
setting.

avirtuos, 2020-01-23:
some awesome ideas in this issue. Would love to see at least some of them (overhang speed, overhang line width, and avoid retracting on overhangs) implemented.

ss89, 2020-07-10:
Is someone actually working on this? Would love to see this feature implemented too, but i'm not into python and c++ deep enough to help out myself.

hitchcme, 2020-12-09:
I think I'm thinking out loud here, but maybe a post-processing script or program to find overhangs and then circular shifts the order of printed segments if the first segment isn't supported by the last layer?  I think the corners can be resolved with the "right" temperature, cooling fan speed, and overhang extrusion speed.

mopedcrosser, 2021-01-31:
Any progress here ? Cura just implemented a slow down on overhangs, its awesome. Fixes curling on steep overhangs and gives nice and smooth walls. Really the only feature missing in prusa slicer that keeps me going back to cura for certain prints

patrickmcquay, 2021-03-09:
Add another vote for this one, it would help me pretty significantly.

I think for my setup specifically, I could get away with a general overhang speed setting (if over x degrees overhang, reduce speed to y) and avoiding seams in overhangs.

VanessaE, 2021-03-09:
A pull request would require code to implement the requested changes.

IMightGitIt, 2021-04-14:
**To me, this is the single most important setting to add in PrusaSlicer.**

This might be the only real thing left in PrusaSlicer, which I cannot fix by changing the current settings.
I mean YES, I could just turn down the speed of external perimeters on the entire print, to avoid the parts with overhang, but that is really inefficient. Being able to have the Prusa Slicer recognize overhang and slice those areas differently would help me so much!

The torture test of this is a pretty basic shape: a cylinder that expands at a 45-degree angle from the centre in both the x and y-direction. This also provokes the seam issue, as there is no good place to place the seam. For any cylinder shape, I usually set the seam to "rear" to at least hide them on a single angle, and make it easier to sand if needed.

The only workaround I currently have is to add a modifier block manually to the areas with an overhang and set the perimeter speed to 50%. This does usually yield much better results, but it is a bit fiddly manual step whenever I spot some overhang.

halfshavedyaks, 2021-09-08:
I would very much like to see such features. I'm not using PrusaSlicer at present, but I would if some of those features were present.

Overhang print speed and overhang fan speed are the most important two settings to add IMO.

kech61, 2023-01-21:
I love the automatic fan speed - but on overhangs I would need fan on.
At least adding the fan speed as option to the modifier would be a great help.

zhangrui1114, 2023-06-02:
Great suggestions, I notice even you set auto-FanSpeed and MAX-Speed to 25, when your model have some overhang, the fan speed still keep MIN-speed automaticely. This is too bad for some filament like ABS/ASA.

closed 2026-09-29 by prusaslicer-bot[bot]