prusa3d/PrusaSlicer #770: Feature request: flow-dependent temperature compensation

VanessaE, 2018-03-07:
### Version

master branch, commit 4a90ab1f6a1390df4ce332b342536ee61db6ebfd

### Operating system type + version

Debian 9.3

### Behavior

When you set up a Print profile for a given speed and layer height, one thing that you must do along with it is set up a corresponding Filament profile with compatible temperatures (and perhaps cooling), since faster speeds require hotter temperatures at the hot end.

This means either creating multiple Filament profiles for each speed, or overriding the print temperature, either of which can be a problem for some users and use-cases.

I think a feature should be added to the Filament Settings that will make it alter the print temperature according to the likely needs of the print.  For example, currently Slic3r has:

```
Temperature (°C)
Extruder:  First Layer: [ 215 ] Other layers: [ 215 ]
Bed:       First Layer: [  60 ] Other layers: [  60 ]
```

What I propose is:

```
Temperature (°C)
Bed:       First Layer: [  60 ] Other layers: [  60 ]

Hot end:
    First Layer, base temperature: [ 210 ]°C  Other layers: [ 205 ]°C
    Temperature compensation: add [ 5 ]°C per 1 mm³/sec above [ 1 ] mm³/sec.
```

Note the renaming to "hot end", as extruders don't melt the plastic, hot ends do, at least in a 3d printer. :stuck_out_tongue:

I don't propose to vary the print temperature in realtime.  Rather, two compensation amounts would be calculated:  the average flow rate of layer 1, and the average for the rest of the print, counting only infill and perimeters (disregarding bridges, gap fill, and non-printing moves, because they'll distort the average).  Those two figures would just be added to the base temperatures and used for the whole print.

In the example above, assuming I have the flow math correct, a speed of 30 mm/s with 0.2 mm layers and a 0.4 mm line width would have a flow rate of ~0.75 mm³/s, so being less than the "above 1 mm³/s" threshold, it would keep the 205° base temperature.  At 60 mm/s the flow rate is ~1.51 mm³/s, so the print would be ~7.5° hotter, at ~212.5°.  At 120 mm/s, you get ~3.02 mm³/s, so the temperature would increase by ~15.1°, to ~220.1°.

Originally I had the idea to only vary with speed, but that didn't account for layer thickness or line width.  This does.

The math could of course allow for too high of a temperature, but let's let the user's firmware cap it.

Some firmwares like Marlin offer auto-temperature, but that produces wildly-fluctuating temperatures on some prints, which may not work out well if the hotend or filament can't react to such changes fast enough (not to mention it being firmware specific).

closed 2019-11-21 by VanessaE

wavexx, 2019-11-21:
Why was this closed?

VanessaE, 2019-11-21:
I figured it isn't needed.  Marlin can do this, too.