prusa3d/PrusaSlicer #772: Outer perimeters of bridges not properly anchored

VanessaE, 2018-03-07:
### Version

master branch, commit 4a90ab1f6a1390df4ce332b342536ee61db6ebfd

### Operating system type + version

Debian 9.3

### Behavior

As it's laying out the perimeters of a bridge, Slic3r starts the thick parts too far from their anchor points.  Each strand of the bridge starts with a very short length from the end of what looks like the normal extrusion rate for a perimeter, then it switches to the bridging flow ratio, and then eventually back to the perimeter flow rate just before reaching the other anchor point.  Both lengths of perimeter-flow-rate-in-the-bridge are a few tenths of a millimeter long, so you get a thick strand with weak anchors, causing uneven stretching of the bridge, leading to drooping and often breaking at the starting end (at least in the case of PLA).  No amount of cooling or speed control has been enough to completely work around this. 

This only seems to affect perimeters, especially external perimeters, where the difference between bridging flow and perimeter flow is greatest.  Solid infill seems to be mostly fine in this regard.  A thinner overall layer height seems to make the problem even worse, probably because that short anchor gets ever thinner, and causes an increasing disparity with the effective bridging flow rate.

[image]

#### STL/Config (.ZIP) where problem occurs

For a test model, use triffid_hunter's bridging torture test, http://www.thingiverse.com/thing:12925
Config files don't matter here, as it's consistent across all of my print, filament, and printer profiles, but here's my config anyway:  [config-20180307.ini.zip](https://github.com/prusa3d/Slic3r/files/1789210/config-20180307.ini.zip)

[ This issue has been copied over, with minor changes, from upstream alexrj/Slic3r as it affects this fork also ]

bubnikv, 2018-03-14:
I understand the issue, but it sounds like we need another parameter to configure the overlap. Do you have any proposal? I personally would like to avoid adding new parameters and do it somehow automagic, but I don't have any good idea how.

bubnikv, 2018-03-14:
I mean, if there is a gap, than it is a bug and it should be fixed. But I suppose you would like to have a bit of overlap of the bridging flow with the supporting layer.

VanessaE, 2018-03-14:
I figured it would be fine to have just a tiny, automatically-set overlap.  Maybe ¼ of the nozzle diameter or so.

VanessaE, 2018-03-14:
After all, bridging infill overlaps 100% over its anchor edges, and it doesn't seem to be a problem.

antoneliasson, 2018-07-25:
I can reproduce this in Slic3r PE 1.39.2 but not in 1.40.1 so it seems fixed. Can you please retest?