prusa3d/PrusaSlicer #777: Rotation Value for X and Y not saved

Saij, 2018-03-08:
### Version
Version 1.39.1-prusa3d-win64

### Operating system type + version
Windows 10 Home 64 Bit
Version 1709

### Behavior
When you rotate an object around Z-axis the value is saved and will be displayed and used again if you rotate again around Z-axis.
But if you do this around X-axis or Y-axis, the value is not saved and you'll always start with a value of 0 degree.

bubnikv, 2018-03-14:
Slic3r internally applies the rotation around X or Y axis directly over the coordinates of the mesh, so there is no data to go be able to rotate back to zero. This may be reworked in Slic3r PE 1.41.xx

bubnikv, 2018-12-11:
We have refactored Slic3r to maintain the rotations around X,Y, non-uniform scaling etc outside of the mesh, and these rotation and scaling values are stored into 3MF as well. It will be part of the Slic3r 1.42 alpha, which is to be released soon. Closing.

closed 2018-12-11 by bubnikv