prusa3d/PrusaSlicer #778: Can modified name of generic object in setting

ghostmod, 2018-03-08:
Hello, i think i have a good idea for next release.

We can not edit the name of new create generic object in setting.
It would be better for the organization to be able to change the name of the generic element.

Thanks you in advance.

[image]

bubnikv, 2018-04-18:
This is something to be possibly implemented in 1.41, where we plan to rewrite this piece of code from Perl into C++.

bubnikv, 2019-01-29:
This works in 1.42.0-alpha4 by editing the names in place in the object list.
https://github.com/prusa3d/slic3r/releases

(Currently it does not work on OSX in place, and we will add a pop up dialog to be open from a pop up menu). Closing.

closed 2019-01-29 by bubnikv