prusa3d/PrusaSlicer #781: Compiling Slic3r from source

Sebastianv650, 2018-03-10:
I spent the last two days finding out out to compile Slic3r from source. The wiki seams to be outdated, for example my last tries failed due to missing curl, once it's now installed it just fails at the next point. It's quite anouing as I was used to play aroung with the source code and PRs in the time back when Slic3r PE was started, now I'm not longer able to do that.

I tried the strawberry perl way first, but I have the feeling the Visual Studio 2013 way is closer to a point which could be called working so I would be happy if someone could do a writeup for this way, please also point out differences (if any) between building for 32 or 64bit.

I found some scattered informations in various issue threads, but nothing realy helpful for getting the full thing working.

Any volunteer?

dartrax, 2018-03-10:
Hello Sebastian!
I also had a hard time trying to get this thing compile a couple of weeks ago.
While I finally succeeded (Visual Studio 2013, 64bit), today I noticed that it is broken again, so I'm at the same boat.

So, I had to download CURL too, that seems to be missing. [In addition to the TTB-Path](https://github.com/prusa3d/Slic3r/issues/679) I also needed to add the CURL-Path to CMakeLists.txt like this:

```
set(TBB_INCLUDE_DIR "C:/local/tbb/include")
set(TBB_LIBRARY "C:/local/tbb/lib/intel64/vc12")
set(CURL_INCLUDE_DIR "C:/local/curl-7.58.0/IA64")
set(CURL_LIBRARY "C:/local/curl-7.58.0/IA64")
```
(assuming you extracted curl into C:\local\curl-7.58.0\)

Where exactly are you hanging right now?

Sebastianv650, 2018-03-10:
> While I finally succeeded (Visual Studio 2013, 64bit)

Maybe that's the problem, I only got a 32bit download link but I try to compile using 64bit perl etc. But according to Google this version is only available in 32bit?

I running the cmd line from wperl64d, executing
`cmake .. -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release -DTBB_LIBRARIES=C:\local\tbb\lib\intel64\vc12\tbb_static.lib -DTBB_LIBRARIES_RELEASE=C:\local\tbb\lib\intel64\vc12\tbb_static.lib -DBoost_LIBRARY_DIRS=C:\local\boost_1_63_0\lib64-msvc-12.0 -DTBB_INCLUDE_DIRS=C:\local\tbb\include -DCURL_LIBRARY=C:\local\curl-7.58.0\lib -DCURL_INCLUDE_DIR=C:\local\curl-7.58.0\include`

I added your set() commands, changing C:/local/curl-7.58.0/IA64 to C:/local/curl-7.58.0/lib as I don't have a IA64 folder? The result is:
-- Found TBB: C:/local/tbb/include (found version "2017.0")
CMake Error at cmake/modules/FindTBB.cmake:284 (set_target_properties):
  set_target_properties called with incorrect number of arguments.
Call Stack (most recent call first):
  xs/CMakeLists.txt:504 (find_package)

--   TBB_INCLUDE_DIRS        = C:/local/tbb/include
--   TBB_DEFINITIONS         =
--   TBB_LIBRARIES           = C:\local\tbb\lib\intel64\vc12\tbb_static.lib;C:/local/tbb/lib/intel64/vc12/tbb_static.lib
--   TBB_DEFINITIONS_DEBUG   = -DTBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = C:/local/tbb/lib/intel64/vc12/tbb_static_debug.lib
--   TBB_DEFINITIONS_RELEASE =
--   TBB_LIBRARIES_RELEASE   = C:\local\tbb\lib\intel64\vc12\tbb_static.lib;C:/local/tbb/lib/intel64/vc12/tbb_static.lib
--   AlienWx_FIND_COMPONENTS=base;core;adv
compilet-X9Zn5.c
   Creating library compilet.lib and object compilet.exp
--   AlienWx_VERSION           = 3.000003
--   AlienWx_INCLUDE_DIRS      = C:\wperl64d\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib;C:\wperl64d\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\include
--   AlienWx_LIBRARIES         = -LIBPATH:C:\wperl64d\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib;wxmsw30u_core.lib;wxbase30u.lib;msvcprt.lib;wxbase30u.lib;wxmsw30u_core.lib;wxmsw30u_adv.lib
--   AlienWx_LIBRARY_DIRS      =
--   AlienWx_DEFINITIONS       = -DWIN32;-D_CRT_SECURE_NO_DEPRECATE=1;-D_CRT_NON_CONFORMING_SWPRINTFS=1;-D_SCL_SECURE_NO_WARNINGS=1;-D__NO_VC_CRTDBG__;-D__WXMSW__;-DNDEBUG;-D_UNICODE;-DWXUSINGDLL;-D_WINDOWS;-DNOPCH
--   AlienWx_DEFINITIONS_DEBUG =
--   AlienWx_CXX_FLAGS         = -GF;-TP;/c;/nologo;/TP;/MD;/Zi;/O2;/W4;/GR;/EHsc
-- Found AlienWx: C:\wperl64d\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib;C:\wperl64d\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\include (found version "3.000003")
-- Found CURL: C:/local/curl-7.58.0/lib (found version "7.58.0")
--   Curl libraries:        = C:/local/curl-7.58.0/lib
--   Curl include dirs:     = C:/local/curl-7.58.0/include
-- Configuring incomplete, errors occurred!

dartrax, 2018-03-10:
By Visual Studio 2013, 64bit I meant, that I use VS 2013 on a 64bit machine ;-) Sorry for not being clear.

As for CURL, I downloaded the compiled version from [Stefan Kanthak](https://skanthak.homepage.t-online.de/download/curl-7.58.0.cab). For my architecture, I needed to set CURL_LIBRARY to "C:/local/curl-7.58.0/AMD64/LIBCURL.LIB", which is included in this download (IA64 did not work, I corrected that above). I also needed the include/curl-directory, to get that I downloaded the Source Archives of libcurl and added that directory to C:/local/curl.../ folder.

The problem you seem to have right now is with the TBB dependency. Do you have a "tbb"-folder at C:\local\tbb ?

I compiled successfully right now, but I'm afraid I real live is calling and I may get back to you in the evening.

Sebastianv650, 2018-03-10:
Yes, tbb is in place. After starting with a new copy of Slic3r sources and with your sets, I was able to continue until 98% of nmake command. Curl again, I will test your curl version.

Creating library XS.lib and object XS.exp
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_formadd referenced in function "public: class Slic3r::Http & __cdecl Slic3r::Http::form_add(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?form_add@Http@Slic3r@@QEAAAEAV12@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@0@Z)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_formfree referenced in function "public: __cdecl Slic3r::Http::priv::~priv(void)" (??1priv@Http@Slic3r@@QEAA@XZ)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_slist_append referenced in function "public: class Slic3r::Http & __cdecl Slic3r::Http::header(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?header@Http@Slic3r@@QEAAAEAV12@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV34@@Z)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_slist_free_all referenced in function "public: __cdecl Slic3r::Http::priv::~priv(void)" (??1priv@Http@Slic3r@@QEAA@XZ)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_easy_strerror referenced in function "public: void __cdecl Slic3r::Http::priv::http_perform(void)" (?http_perform@priv@Http@Slic3r@@QEAAXXZ)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_easy_init referenced in function "public: __cdecl Slic3r::Http::priv::priv(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (??0priv@Http@Slic3r@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_easy_setopt referenced in function "public: static class Slic3r::Http __cdecl Slic3r::Http::post(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >)" (?post@Http@Slic3r@@SA?AV12@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_easy_perform referenced in function "public: void __cdecl Slic3r::Http::priv::http_perform(void)" (?http_perform@priv@Http@Slic3r@@QEAAXXZ)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_easy_cleanup referenced in function "public: __cdecl Slic3r::Http::priv::~priv(void)" (??1priv@Http@Slic3r@@QEAA@XZ)
libslic3r_gui.lib(Http.cpp.obj) : error LNK2019: unresolved external symbol curl_easy_getinfo referenced in function "public: void __cdecl Slic3r::Http::priv::http_perform(void)" (?http_perform@priv@Http@Slic3r@@QEAAXXZ)
XS.dll : fatal error LNK1120: 10 unresolved externals
NMAKE : fatal error U1077: '"C:\Program Files\CMake\bin\cmake.exe"' : return code '0xffffffff'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.

Sebastianv650, 2018-03-10:
Using your curl version make the build process sucessful, but the test using ctest --verbose fails:

UpdateCTestConfiguration  from :C:/Users/Sebastian/Desktop/Slic3r-master/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :C:/Users/Sebastian/Desktop/Slic3r-master/build/DartConfiguration.tcl
Test project C:/Users/Sebastian/Desktop/Slic3r-master/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: xs

1: Test command: C:\wperl64d\bin\perl.exe "C:/wperl64d/bin/prove" "-I" "C:/Users/Sebastian/Desktop/Slic3r-master/local-lib/lib/perl5"
1: Test timeout computed to be: 9.99988e+06
1: Can't open perl script "C:/wperl64d/bin/prove": No such file or directory
1/2 Test #1: xs ...............................***Failed    0.06 sec
test 2
    Start 2: integration

2: Test command: C:\wperl64d\bin\perl.exe "C:/wperl64d/bin/prove"
2: Test timeout computed to be: 9.99988e+06
2: Can't open perl script "C:/wperl64d/bin/prove": No such file or directory
2/2 Test #2: integration ......................***Failed    0.02 sec

0% tests passed, 2 tests failed out of 2

starting slic3r fails also:

Can't load 'C:\Users\SEBAST~1\Desktop\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread/auto/Slic3r/XS/XS.dll' for module Slic3r::XS: load_file:Das angegebene Modul wurde nicht gefunden at c:/wperl64d/lib/XSLoader.pm line 71.
  at C:\Users\SEBAST~1\Desktop\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread/Slic3r/XS.pm line 22.
Compilation failed in require at C:/Users/Sebastian/Desktop/Slic3r-master/lib/Slic3r.pm line 50.
BEGIN failed--compilation aborted at C:/Users/Sebastian/Desktop/Slic3r-master/lib/Slic3r.pm line 50.
Compilation failed in require at slic3r.pl line 16.
BEGIN failed--compilation aborted at slic3r.pl line 16.

dartrax, 2018-03-10:
I did not tried the test after it failed once a couple of weeks ago...

Sebastianv650, 2018-03-10:
I also only tried the test because it didn't start. the error during perl Slic3r.pl doens't make much sense to me, as the XS.dll is inside this path.

dartrax, 2018-03-10:
Just to let you know, I get the same error when try to start slicer.

vojtechkral, 2018-03-12:
@Sebastianv650 @dartrax With the integration of libcurl I added a bunch of improvements to the build system that should make things easier, especially on Windows. There is a new how-to written that should replace the outdated wiki, located in the `doc` subdir. Right now I am waiting for the merge of #771 which should conclude the effort.

If you want to compile Slic3r PE right now before #771 is merged, please checkout the `msvc` branch, have a look at [the how-to](https://github.com/prusa3d/Slic3r/blob/msvc/doc/How%20to%20build%20-%20Windows.md) to see if it helps you. Hopefully it should be quite a bit simpler, at least that was my intention. If not, I'd be happy to receive feedback on it. Thanks!

Once the last fix is merged I should probably llink the how-to from the wiki.

Sebastianv650, 2018-03-12:
Thank you very much @vojtechkral, worked like a charm! Only one note for the how-to page: It's not mentioned that your will also need to download & install cmake. I think this should be added to the list of needed software.

vojtechkral, 2018-03-12:
@Sebastianv650 Ok, great, thanks for testing. And yes, I'll add a CMake mention, that's a good idea...

bubnikv, 2018-03-14:
Great it is working for @Sebastianv650 . Closing.

closed 2018-03-14 by bubnikv

dartrax, 2018-03-16:
Where exactly do I have to unzip the slic3r-destdir-64.7z to?
I now have:
```
C:\local\slic3r-destdir-64\usr\local\lib
C:\local\slic3r-destdir-64\usr\include\boost-1_63\boost
C:\local\slic3r-destdir-64\usr\include\curl
C:\local\slic3r-destdir-64\usr\include\tbb
```
Is that correct? It looks like it can't find anything of that.
I'm getting:
```
CMake Warning at C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  xs/CMakeLists.txt:490 (find_package)

CMake Warning at C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  xs/CMakeLists.txt:490 (find_package)

CMake Warning at C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  xs/CMakeLists.txt:490 (find_package)

CMake Warning at C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  xs/CMakeLists.txt:490 (find_package)

CMake Warning at C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  xs/CMakeLists.txt:490 (find_package)

CMake Warning at C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  xs/CMakeLists.txt:490 (find_package)

CMake Error at C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:1928 (message):
  Unable to find the requested Boost libraries.

  Unable to find the Boost header files.  Please set BOOST_ROOT to the root
  directory containing Boost or BOOST_INCLUDEDIR to the directory containing
  Boost's headers.
Call Stack (most recent call first):
  xs/CMakeLists.txt:490 (find_package)

CMake Error at C:/Program Files/CMake/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find TBB (missing: TBB_INCLUDE_DIRS)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake:273 (find_package_handle_standard_args)
  xs/CMakeLists.txt:508 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/local/Slic3r/build/CMakeFiles/CMakeOutput.log".
See also "C:/local/Slic3r/build/CMakeFiles/CMakeError.log".
```

bubnikv, 2018-03-16:
You have to update the perl distribution according to
https://github.com/prusa3d/Slic3r/blob/master/doc/How%20to%20build%20-%20Windows.md
, or just update C:\wperl64d\cmdline.bat to:

set PATH=%PATH%;c:\wperl64d\site\bin;c:\wperl64d\bin
set PERL_CPANM_HOME=c:\wperl64d\cpanm
set SLIC3R_STATIC=1
set CMAKE_PREFIX_PATH=C:\local\slic3r-destdir-64\usr\local
"C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\vcvarsall.bat" amd64

Important here is the new CMAKE_PREFIX_PATH variable.

dartrax, 2018-03-16:
I already did that. I looked into cmdline_init.bat and noticed the Path C:\local\slic3r-destdir-64\usr\local\. Therefore, I concluded that I have to unzip slic3r-destdir-64.7z in such a way that this path exists.
Still the same, something is missing...

bubnikv, 2018-03-17:
```
C:\local\slic3r-destdir-64\usr\local\lib
C:\local\slic3r-destdir-64\usr\include\boost-1_63\boost
C:\local\slic3r-destdir-64\usr\include\curl
C:\local\slic3r-destdir-64\usr\include\tbb
```

This is not quite correct, it should be 
```
C:\local\slic3r-destdir-64\usr\local\lib
C:\local\slic3r-destdir-64\usr\local\include\boost-1_63\boost
C:\local\slic3r-destdir-64\usr\local\include\curl
C:\local\slic3r-destdir-64\usr\local\include\tbb
```

dartrax, 2018-03-18:
Thanks for spotting that! 
It‘s working now, I can start it in VS2013 and it even stops at breakpoints and I can view variable values. That will help debugging and polishing my preferred-seam-location feature :)

Sebastianv650, 2018-11-25:
Should the doc - How to build be still up to date? I'm asking because I run into errors even if I start with completely fresh folders. Last lines, not shure if it's due to wxwidgets or the compiler "Compiler/MSVC-ASM" thing:

[ 66%] Completed 'googletest-download'
[100%] Built target googletest-download
-- Could NOT find PythonInterp (missing: PYTHON_EXECUTABLE)
CMake Warning at src/CMakeLists.txt:34 (message):
  Using builtin libpng.  This can cause crashes on some platforms.

-- Warning: Did not find file Compiler/MSVC-ASM
-- WXWIN environment set to:
CMake Error at C:/Program Files/CMake/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindwxWidgets.cmake:953 (find_package_handle_standard_args)
  src/CMakeLists.txt:49 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Sebastian/Desktop/Slic3r-master/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Sebastian/Desktop/Slic3r-master/build/CMakeFiles/CMakeError.log".

bubnikv, 2018-11-25:
We switched our master to native only (with the exception to Perl bindings
to libslic3r and the associated unit / integration tests). We are now using
wxWidgets 3.1.1. On Windows, we are compiling wxWidgets statically. There
is a cmakefile in the deps directory, which pulls and compiles all the
dependencies. For wxWidgets to compile statically, you have to apply a
patch, which is part of the deps cmake script. I personally (and our other
2 Windows developers) are compiling wxWidgets 3.1.1 from sources using
visual studio IDE after we apply the patch for static compilation of libpng
with a prefix. You have to set the WXWIN environment variable to point to
your wxWidgets install directory before running cmake.

Please note that by default the current master does not require perl. It is
only if you call cmake with the -DSLIC3R_PERL_XS=on, when perl is required
and the Perl unit / integration tests are executed.

Sebastianv650, 2018-11-26:
Thanks for the information, but I guess I have to wait until the build instructions are updated. Maybe you could add a "not working at the moment" note until then to prevent others from confusion?

I was able to run the deps cmake thing without errors (I guess), but I have no idea where or if the libraries are installed after 30min of searching. This is a little bit over my head without the howto ;)