prusa3d/PrusaSlicer #783: Clicking on "Browse..." under "OctoPrint upload" results in a crash

emreerokyar, 2018-03-11:
### Version
Slic3r Prusa Edition
Version 1.31.2-prusa3d

### Operating system type + version
macOS High Sierra
Version 10.13.2

### Behavior
**Problem description:**
Clicking on "Browse..." under "OctoPrint upload" results in a crash.

**Steps needed to reproduce the problem:**
**1.** Start Slic3r Prusa Edition.
**2.** Go to the "Printer Settings" pane.
**3.** Click on the "Browse..." button under "OctoPrint upload" section.

**Expected results:**
I am expecting a "Browse" window to appear.

**Actual results:**
Slic3r Prusa Edition crashes.

**Is this a new feature request?**
No

#### STL/Config (.ZIP) where problem occurs
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1800905/config.ini.zip)

[image]

bubnikv, 2018-03-12:
Slic3r PE 1.31.2 is very outdated. Would you please try with
https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1 ?

vojtechkral, 2018-03-15:
The upcomming release uses a new backend code for device discovery, and so this is not going to be applicable any more.

closed 2018-03-15 by vojtechkral