prusa3d/PrusaSlicer #785: Feature Request: Date & Time when Print will be finished

dartrax, 2018-03-12:
We have an "Estimated printing time" now which I greatly appreviate.
It would be helpful to have this duration added to current time and date, so you don't have to do this calculation in your head - you would see in one glance, at which time of the day the print will be finished. Our prints often go more than 24 hours, so this can be an error-prone task to do manually.

I suggest either the estimated ending date and time to be added in brackets behind the duration, to be added in a new line or as tooltip when the estimated printing time is hovered.
Of course this is valid only when printing starts right now, but that should be obvious to everyone...

bubnikv, 2018-08-02:
Anyone else founds this request useful? I am inclined to refuse such request to keep the UI uncluttered. Or maybe @YuSanka could implement it as a tooltip?

dartrax, 2018-08-02:
A tooltip would be perfectly enough. 💛

closed 2026-09-29 by prusaslicer-bot[bot]