prusa3d/PrusaSlicer #788: Make recipe fails with "Error 1" on PlaceholderParser.cpp.o

Patronics, 2018-03-13:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
1.40.0.-alpha-198-gdb61c23
### Operating system type + version
_What OS are you using, and state any version #s_
Linux Raspbian Jesse
### Behavior
* _Describe the problem_
Make consistently fails to build Slic3r from source when building the file PlaceholderParser.cpp.o
* _Steps needed to reproduce the problem_

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Make completes sucessfully
* _Actual Results_
I get this error message, and make exits.
[image]

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
N/A

Patronics, 2018-03-13:
Also (almost certainly unrelated to the above problem) although most of the warnings given as make runs are likely unimportant (such as the unused variables), I saw a few where it stated "`trying to compare a signed and unsigned value`" in if statements comparing a variable to -1, which will likely never run due to the unsigned variable never being negative. I don't know the ramifications of this in the particular code, but thought I should mention it, as it likely results in some unexplained behavior or bug.

bubnikv, 2018-03-14:
Would you please post the build log in a text form? I think I don't see the error line, just the warning lines.

Patronics, 2018-03-14:
That’s the thing, there is no clear error message, but the last 10 lines or so of the screenshot are the error message.
Here’s a link to the text form of the log:
https://gist.github.com/Patronics/10db9ff5f6d676742d3ad330cf5be38b
I see that in addition to the error messages at the bottom there’s also 'error: ‘copy’ is not a member of ‘Slic3r::qi’’

bubnikv, 2018-03-14:
Which version of boost? Can you compile an up to date boost? We are using 1.63.

Patronics, 2018-03-17:
OK, after updating boost (and installing ExtUtils::Typemaps and ExtUtils::XSpp based on other error messages), the make completes successfully, however the next step in the directions, running `ctest --verbose` gives the following error: 
`2: /usr/bin/perl: symbol lookup error: /home/pi/Slic3r/t/../local-lib/lib/perl5/arm-linux-gnueabihf-thread-multi-64int/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost16re_detail_10660012perl_matcherIN9__gnu_cxx17__normal_iteratorIPKcSsEESaINS_9sub_matchIS6_EEENS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE14construct_initERKNS_11basic_regexIcSD_EENS_15regex_constants12_match_flagsE
2: t/custom_gcode.t ............... 
2: Dubious, test returned 127 (wstat 32512, 0x7f00)
2: Failed 63/77 subtests `

Here's the link to the complete output: [https://gist.github.com/Patronics/c7be2a5b27cbb8c5ef73c2d638213266](https://gist.github.com/Patronics/c7be2a5b27cbb8c5ef73c2d638213266)

how can I fix this?

bubnikv, 2018-03-17:
> c7be2a5b27cbb8c5ef73c2d638213266

Seemingly a wrong boost library is being loaded. You may try your luck with
compiling the dependencies statically. We currently use the following
process on our Linux build server to collect and compile static
dependencies:

https://github.com/prusa3d/Slic3r/blob/master/doc/deps-build/unix-static/Makefile

Don't forget to set
export SLIC3R_STATIC=1
export CMAKE_PREFIX_PATH=$DESTDIR/usr/local
when building the dependencies statically.

If you get it working, it would be great if you could create a Wiki page :-)

Patronics, 2018-03-17:
I managed to fix the problem, and didn’t need to compile the dependencies statically. It turned out the problem was that since make had succeeded in compiling around half the files before getting to the one it failed at, once I updated boost to 1.66 and ran make again it didn’t recompile the already made files from the old boost version. Deleting the build directory and redoing the procedure from running Build.PL fixed the problem 😄. Thanks for all the help 😀. I could try to create a wiki page, but it’d probably end up just being how to deal with my exact problem 😜. Summary of solution: update boost, make sure you have all dependencies, and make sure make starts from the beginning.