prusa3d/PrusaSlicer #790: NVIDIA OpenGL Driver kernel exception after hibernate

dartrax, 2018-03-13:
### Version
Slic3r-1.39.1-beta-win64-full-20180219

### Operating system type + version
Win 10 64bit

### Behavior
I've got a new PC and recovered from hibernating mode. Slic3r greeted me with the following Error, that I will put here for reference in case somebody else has such a problem or it is already known.

[image]

```
Slic3r Prusa Edition
Version: 1.39.0
Build:   UNKNOWN
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.16299.248]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

  @INC:
    C:\Users\traxel\Downloads\Slic3r-1.39.1-beta-dartrax-win64-full-20180219\local-lib\lib\perl5\5.24.0\MSWin32-x64-multi-thread
    C:\Users\traxel\Downloads\Slic3r-1.39.1-beta-dartrax-win64-full-20180219\local-lib\lib\perl5\5.24.0
    C:\Users\traxel\Downloads\Slic3r-1.39.1-beta-dartrax-win64-full-20180219\local-lib\lib\perl5\MSWin32-x64-multi-thread
    C:\Users\traxel\Downloads\Slic3r-1.39.1-beta-dartrax-win64-full-20180219\local-lib\lib\perl5
    C:/Users/traxel/Downloads/Slic3r-1.39.1-beta-dartrax-win64-full-20180219/lib
    .
OpenGL installation
  Using POGL v0.6704
  GL version:   4.6.0 NVIDIA 390.77
  vendor:       NVIDIA Corporation
  renderer:     Quadro M2000/PCIe/SSE2
  GLSL version: 4.60 NVIDIA
Installed extensions (* implemented in the module):
  GL_AMD_multi_draw_indirect
  GL_AMD_seamless_cubemap_per_texture
  GL_AMD_vertex_shader_layer
  GL_AMD_vertex_shader_viewport_index
  GL_ARB_ES2_compatibility
  GL_ARB_ES3_1_compatibility
[... 345 lines cut ...]
  GL_NV_transform_feedback
  GL_NV_transform_feedback2
  GL_NV_uniform_buffer_unified_memory
  GL_NV_vertex_array_range
  GL_NV_vertex_array_range2
  GL_NV_vertex_attrib_integer_64bit
  GL_NV_vertex_buffer_unified_memory
  GL_NV_vertex_program
  GL_NV_vertex_program1_1
    Depends on NV_vertex_program
  GL_NV_vertex_program2
    Depends on NV_vertex_program
* GL_NV_vertex_program2_option
* GL_NV_vertex_program3
  GL_NV_viewport_array2
  GL_NV_viewport_swizzle
* GL_S3_s3tc
* GL_SGIS_generate_mipmap
* GL_SGIS_texture_lod
* GL_SGIX_depth_texture
* GL_SGIX_shadow
* GL_SUN_slice_accum
  GL_WIN_swap_hint
  WGL_EXT_swap_control
```

bubnikv, 2018-03-14:
Sorry, but if Inventor crashed, then I suppose it is a driver issue specific to your hardware setup. I don't think it helps to keep your issue open. Closing.

closed 2018-03-14 by bubnikv

rongith, 2018-03-20:
The same here with Solidworks, nVidia M4000 and slic3r. I need to see what happens next time the issue occur.