prusa3d/PrusaSlicer #791: Output FileFormat

nerocide, 2018-03-13:
### Version
I compiled Slic3rPE v1.39 from repo (tag version_1.40.0-alpha-196-g83a2d2af)

### Operating system type + version
Linux kali 4.14.0-kali3-amd64 #1 SMP Debian 4.14.17-1kali1 (2018-02-16) x86_64 GNU/Linux

### Behavior
* at least the following variable names can't be used in **output filename forrmat** [print_preset] [filament_preset] [printer_preset]

* _Steps needed to reproduce the problem_

In print settings > Output Options > Output filename format add the following string **[input_filename_base]_[print_preset]_[filament_preset]_[printer_preset].gcode**
Slice your STL and export as gcode.

This will raise the following error
[image]

Any ideas about this behaviour?

jneilliii, 2018-03-23:
I can confirm I have this issue as well with latest 1.39.1 windows x64 release.

[image]

jneilliii, 2018-03-24:
I just did a little testing and by changing the name, saving the print settings, and then putting it back and saving the print settings again, it appears to be working again. Until I restart the program.  Once I restart the program the error returns.

jneilliii, 2018-03-24:
Some additional information.  This seems to only be an issue if there are more than one filename tokens.  If you only have 1, then restarting the application does not make the error reoccur.

jneilliii, 2018-03-24:
Acually looks like it might be a specific issue with [filament_preset} token.  I tried a couple of different ones, like [input_filename_base]_[hour]_[minute].gcode and that works without issue as well.

nerocide, 2018-04-07:
As far as my last build from **master branch** (**version_1.40.0-alpha-285-gc6623bb2**), it works like a charm now.

Any confirmation?

bubnikv, 2018-05-16:
I was able to reproduce it.

bubnikv, 2018-05-16:
Fixed with https://github.com/prusa3d/Slic3r/commit/0b4a61a8bad7d507a6ad93314346a879682a2b5c
It will be part of the upcoming beta.
Closing, please reopen if you find out it does not work with the 1.40.0-beta. Thanks.

closed 2018-05-16 by bubnikv

nerocide, 2018-05-16:
Thanks @bubnikv, just compiled **version_1.40.0-alpha1-9-g6d98c2b1** and it works as expected.