prusa3d/PrusaSlicer #792: Error on slicing model

luckyguido, 2018-03-14:
### Version
1.39.1-prusa3d-win64

### Operating system type + version
Windows 10 x64

### Behavior
Slicing a model the output miss some internal hole, on 1.38 this doesn't happen
attached images for the 2 version results.

#### STL/Config (.ZIP) where problem occurs
printed model is https://www.thingiverse.com/thing:1212828
settings are default bundled one with 0.15mm layer height

[image]
[image]

bubnikv, 2018-03-14:
@FidelCapo Would you please verify this issue with our latest alpha? Thanks.

govido, 2018-03-18:
I have also problems with slicing errors in the new stable build 1.39. 1.37 and 1.38 worked more reliably. Is there already a precompiled alpha or is self-compiling of version 1.40 needed?

bubnikv, 2018-03-18:
> I have also problems with slicing errors in the new stable build 1.39. 1.37 and 1.38 worked more reliably.

Would you please post the models?

Some of the fixes we did may break other models. Luckily for 1.40 we are integrating some more intelligence to work around a wider set of model errors.

govido, 2018-03-18:
[image]

[extruder_test.zip](https://github.com/prusa3d/Slic3r/files/1822650/extruder_test.zip)

Hope this helps

bubnikv, 2018-03-18:
@govido
It looks to me with the current alpha. More test cases are appreciated.

govido, 2018-03-18:
[image]
Found another model from an abandoned project with similar behavior. 
[carriage module test.zip](https://github.com/prusa3d/Slic3r/files/1822672/carriage.module.test.zip)

bubnikv, 2018-03-18:
So far so good with the latest alpha.

jgastonraoul, 2018-03-30:
I've some sort of similar problem with a model...
[image]

@bubnikv Can you tell me if the problem is the same with the latest alpha ?

[poignee droite27.stl.zip](https://github.com/prusa3d/Slic3r/files/1863012/poignee.droite27.stl.zip)

FidelCapo, 2018-04-03:
@jgastonraoul This model sliced ok with the latest alpha.

jgastonraoul, 2018-04-03:
@FidelCapo Thanks for the test.
Is there a way to download this alpha (for Mac) somewhere or do I have to build it by myself ?

bubnikv, 2018-04-18:
@FidelCapo Please add this model to our test database.
This issue will be fixed in the upcoming 1.40 beta. Closing.

closed 2018-04-18 by bubnikv

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon.