prusa3d/PrusaSlicer #794: Print size not same as file size

jimbo777, 2018-03-14:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-03-14:
Sorry, but you did not really provide much useful information. The only thing we got is the title

> Print size not same as file size

and I have no idea, what you mean by that.

jimbo777, 2018-03-14:
sorry - computer error

I am attaching the suggested files:
[x-carriage-back.zip](https://github.com/prusa3d/Slic3r/files/1812381/x-carriage-back.zip)

Slic3r 1.39.1  
windows 10 with latest updates
the printed object does not conform to the .stl file in dimensions - this is only internal dimensions, the external measurements are ok
I have tried this several times with exactly the same result, some with other files and test objects I created with Tinker cad

the file "carriage-back.stl   gave these results:
                                     print size          stl file dimensions
width widest point          52mm                 52mm
length longest                72.2mm              73.5  
internal hole                     5.65mm              6mm
internal hole                     2.5mm                3.2mm
internal oval short side     9.6mm              10mm

the internal dimensions are consistently smaller than the file indicates they should be!

bubnikv, 2018-03-14:
The outer dimensions should be quite precise. If they are not, then your printer is not correctly calibrated. What kind of printer are you printing on?

The FDM technology is known to shrink small holes due to the internal stresses left in the filament extruded by the tiny nozzle orifice.

jimbo777, 2018-03-14:
sorry again, Original Prusa MK2 about 14 months old. Has been working excellent. Actually the outside dimensions are very exact, I have also tested with a calibration cube. I am concerned with the inside dimensions. They are small and the upgraded parts I printed may not work. I have printed a part for the steel rod which should measure 8mm but printed to 7.87 and there was no way to make it fit.

what should I do?

bubnikv, 2018-03-14:
I suppose this is not a new issue. You may try to lower the extrusion multiplier by a couple of percent, but in general one needs to just model the holes a bit bigger for FDM.

Your issue is actually a duplicate of https://github.com/prusa3d/Slic3r/issues/515
and there is a lot of information available at https://github.com/prusa3d/Slic3r/issues/515.

bubnikv, 2018-08-02:
Duplicate of #515. Closing.

closed 2018-08-02 by bubnikv