prusa3d/PrusaSlicer #797: Feature Request: Fan on for n layers above overhang/bridges.

Squid116, 2018-03-15:
Printing in PETG I have observed that the a bridge will print perfectly due to the bridging fan settings, however when it prints the next layer the bridge begins to sag - due to the heat retention and transfer properties of PETG.  You can get around this and print nice bridges by having the fan always on, but that impacts the rest of the part's strength - so is not ideal.  If there was an option to have the fan on at bridging speed for a configurable number of layers above a bridge/overhang that would make things easier when printing PETG (and probably some other filaments).

govido, 2018-03-18:
+1
I think this would be especially great for printing very thin layers beyond 0.15mm. these overhangs tend to curl an sag down very often. Also an option to spin the fan x-seconds in advance would be great. Depending on how long the fan needs to get up to 100% speed.

mavu, 2018-05-19:
I was going to suggest a similar change, for small overhangs, which often tend to curl up, but print better when slowing the print down to 60%.

Some additional handling of overhangs in edge cases ( ha! :) ) would be nice. 

Automatic fan on small overhangs, bridging fan on later layers as suggested above , etc.

francisco-pereira, 2019-06-20:
+1 Spin the fan a few seconds in advance.
Also, an option to select on which layers the fan should be on would be great. (something similar as it's now done with layer heights)

closed 2026-09-29 by prusaslicer-bot[bot]