prusa3d/PrusaSlicer #799: 3D render window running slowly after windows update

BlazingAngel665, 2018-03-17:
### Version
1.386.6-win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10.0.16299.309

### Behavior
* _Describe the problem_
When the software opens, it runs very slowly as long as any of the 3D render window tabs are selected. The program runs slowly regardless of which, if any, .stl file is loaded. All program functions still work.
* _Steps needed to reproduce the problem_
A windows update was installed just prior to observing this behavior. Reverting to OpenGL 1.1 rendering did not resolve the issue.
The program was run via shortcut on the desktop. The program is immediately sluggish. After clicking the 2D tab and waiting 10-30 seconds the program runs at normal speed. Upon returning to the 3D tab the program runs slowly again.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Previous program functionality
* _Actual Results_
 
[image]
About 1 minute after attempting to load a model
[image]
Immediately after attempting to pan the camera. Icons disappear the and GUI lags for 30 ish seconds.

_Is this a new feature request?_
This is a bug report

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
Occurs on program opening

#### System Information Dump
Slic3r Prusa Edition
Version: 1.38.6-prusa3d-win64
Build:   Slic3r-1.38.6-prusa3d-win64-full-201712292042
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.16299.309]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

  @INC:
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0\MSWin32-x64-multi-thread
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\MSWin32-x64-multi-thread
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5
    C:/PROGRA~1/Prusa3D/Slic3rPE/lib
    C:/Program Files/Prusa3D/Slic3rPE/lib
    .
OpenGL installation
  Using POGL v0.6704
  GL version:   4.0.0 - Build 10.18.10.4358
  vendor:       Intel
  renderer:     Intel(R) HD Graphics 4000
  GLSL version: 4.00 - Build 10.18.10.4358
Installed extensions (* implemented in the module):
  GL_3DFX_texture_compression_FXT1
  GL_ARB_ES2_compatibility
  GL_ARB_ES3_compatibility
  GL_ARB_arrays_of_arrays
  GL_ARB_base_instance
  GL_ARB_blend_func_extended
  GL_ARB_buffer_storage
* GL_ARB_color_buffer_float
  GL_ARB_compatibility
  GL_ARB_compressed_texture_pixel_storage
  GL_ARB_conservative_depth
  GL_ARB_copy_buffer
  GL_ARB_debug_output
  GL_ARB_depth_buffer_float
  GL_ARB_depth_clamp
* GL_ARB_depth_texture
* GL_ARB_draw_buffers
  GL_ARB_draw_buffers_blend
  GL_ARB_draw_elements_base_vertex
  GL_ARB_draw_indirect
  GL_ARB_draw_instanced
  GL_ARB_explicit_attrib_location
  GL_ARB_fragment_coord_conventions
* GL_ARB_fragment_program
* GL_ARB_fragment_program_shadow
* GL_ARB_fragment_shader
  GL_ARB_framebuffer_no_attachments
  GL_ARB_framebuffer_object
  GL_ARB_framebuffer_sRGB
  GL_ARB_geometry_shader4
  GL_ARB_get_program_binary
  GL_ARB_gpu_shader5
  GL_ARB_gpu_shader_fp64
* GL_ARB_half_float_pixel
  GL_ARB_half_float_vertex
  GL_ARB_instanced_arrays
  GL_ARB_internalformat_query
  GL_ARB_internalformat_query2
  GL_ARB_map_buffer_alignment
  GL_ARB_map_buffer_range
  GL_ARB_multi_draw_indirect
* GL_ARB_multisample
* GL_ARB_multitexture
  GL_ARB_occlusion_query
  GL_ARB_occlusion_query2
* GL_ARB_pixel_buffer_object
* GL_ARB_point_parameters
* GL_ARB_point_sprite
  GL_ARB_program_interface_query
  GL_ARB_provoking_vertex
  GL_ARB_robustness
  GL_ARB_sample_shading
  GL_ARB_sampler_objects
  GL_ARB_seamless_cube_map
  GL_ARB_separate_shader_objects
  GL_ARB_shader_atomic_counters
  GL_ARB_shader_bit_encoding
* GL_ARB_shader_objects
  GL_ARB_shader_precision
  GL_ARB_shader_subroutine
* GL_ARB_shading_language_100
  GL_ARB_shading_language_420pack
  GL_ARB_shading_language_packing
* GL_ARB_shadow
  GL_ARB_stencil_texturing
  GL_ARB_sync
  GL_ARB_tessellation_shader
* GL_ARB_texture_border_clamp
  GL_ARB_texture_buffer_object_rgb32
  GL_ARB_texture_buffer_range
  GL_ARB_texture_compression
  GL_ARB_texture_compression_bptc
  GL_ARB_texture_compression_rgtc
* GL_ARB_texture_cube_map
  GL_ARB_texture_cube_map_array
* GL_ARB_texture_env_add
* GL_ARB_texture_env_combine
  GL_ARB_texture_env_crossbar
* GL_ARB_texture_env_dot3
* GL_ARB_texture_float
  GL_ARB_texture_gather
  GL_ARB_texture_multisample
* GL_ARB_texture_non_power_of_two
  GL_ARB_texture_query_lod
* GL_ARB_texture_rectangle
  GL_ARB_texture_rg
  GL_ARB_texture_rgb10_a2ui
  GL_ARB_texture_storage
  GL_ARB_texture_storage_multisample
  GL_ARB_texture_swizzle
  GL_ARB_timer_query
  GL_ARB_transform_feedback2
  GL_ARB_transform_feedback3
  GL_ARB_transform_feedback_instanced
  GL_ARB_transpose_matrix
  GL_ARB_uniform_buffer_object
  GL_ARB_vertex_array_bgra
  GL_ARB_vertex_array_object
  GL_ARB_vertex_attrib_64bit
  GL_ARB_vertex_attrib_binding
* GL_ARB_vertex_buffer_object
* GL_ARB_vertex_program
* GL_ARB_vertex_shader
  GL_ARB_vertex_type_2_10_10_10_rev
  GL_ARB_viewport_array
  GL_ARB_window_pos
  GL_ATI_separate_stencil
* GL_EXT_abgr
* GL_EXT_bgra
* GL_EXT_blend_color
  GL_EXT_blend_equation_separate
  GL_EXT_blend_func_separate
  GL_EXT_blend_minmax
* GL_EXT_blend_subtract
  GL_EXT_clip_volume_hint
  GL_EXT_compiled_vertex_array
  GL_EXT_draw_buffers2
  GL_EXT_draw_range_elements
  GL_EXT_fog_coord
  GL_EXT_framebuffer_blit
  GL_EXT_framebuffer_multisample
* GL_EXT_framebuffer_object
  GL_EXT_geometry_shader4
  GL_EXT_gpu_program_parameters
  GL_EXT_gpu_shader4
  GL_EXT_multi_draw_arrays
  GL_EXT_packed_depth_stencil
  GL_EXT_packed_float
* GL_EXT_packed_pixels
* GL_EXT_rescale_normal
  GL_EXT_secondary_color
* GL_EXT_separate_specular_color
  GL_EXT_shader_integer_mix
* GL_EXT_shadow_funcs
  GL_EXT_stencil_two_side
* GL_EXT_stencil_wrap
* GL_EXT_texture3D
  GL_EXT_texture_array
  GL_EXT_texture_buffer
  GL_EXT_texture_compression_s3tc
  GL_EXT_texture_edge_clamp
  GL_EXT_texture_env_add
* GL_EXT_texture_env_combine
* GL_EXT_texture_filter_anisotropic
  GL_EXT_texture_integer
* GL_EXT_texture_lod_bias
  GL_EXT_texture_rectangle
  GL_EXT_texture_sRGB
  GL_EXT_texture_sRGB_decode
  GL_EXT_texture_shared_exponent
  GL_EXT_texture_snorm
  GL_EXT_texture_storage
  GL_EXT_texture_swizzle
  GL_EXT_transform_feedback
  GL_IBM_texture_mirrored_repeat
  GL_INTEL_map_texture
  GL_INTEL_performance_queries
  GL_INTEL_performance_query
  GL_KHR_blend_equation_advanced
  GL_KHR_debug
* GL_NV_blend_square
  GL_NV_conditional_render
  GL_NV_primitive_restart
* GL_NV_texgen_reflection
* GL_SGIS_generate_mipmap
  GL_SGIS_texture_edge_clamp
* GL_SGIS_texture_lod
  GL_SUN_multi_draw_arrays
  GL_WIN_swap_hint

bubnikv, 2018-03-17:
This issue has been worked around in the last official release.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1

Please report whether the Slic3r update helped. Thank you,.

bubnikv, 2018-04-18:
No response, and likely this issue has been fixed with 1.39.1. Closing.

closed 2018-04-18 by bubnikv