prusa3d/PrusaSlicer #800: Feature Request: Show the estimated time of a print when slicing, before save/export

Patronics, 2018-03-18:
### Version
N/A

### Behavior
* _Describe the problem_
When slicing a file, it Slic3r does not show the estimate of how long the print will take until you export the gcode. 
* _Steps needed to reproduce the problem_
Slice a model
* _Expected Results_
Slic3r displays the estimated duration of the print with current settings when the "Slice now" button is pressed
* _Actual Results_
Slic3r does not show the estimated print duration until after the gcode is exported

_Is this a new feature request?_
Yes
#### STL/Config (.ZIP) where problem occurs
Any

daktylus, 2018-03-19:
+1  !!

It is annoying if you want to tweak the print time and have to save the gcode every time to see the effect of your changes.

Simple Solution: This would simply be possible with saving the gcode in a temporary file every time it is sliced.

triplepoint, 2018-03-20:
There was a discussion at some point of encoding the print time and material usage estimates into a structured-data comment at the top of the exported gcode, such that it could be read by an octoprint plugin or the printer itself to display more accurate estimates.  

It seems like if we generated these estimates before exporting the gcode file, it would be a lot easier to insert these comments.

This would also solve the "I forgot to write down slic3r's estimate and now it's gone" problem - you could just open the gcode file and examine the comment.

PaulKlinger, 2018-03-20:
The estimates are already included in the gcode, at the end of the file.

sej7278, 2018-04-16:
I noticed it doesn't even show the time when you export the gcode - you have to export it a second time for it to work. Its always in the file though - is it reading the value back from the file or something stupid?

Debian Sid.

bubnikv, 2018-04-18:
We will implement automatic export of G-code in 1.41. Unfortunately it will not be part of the upcoming 1.40 beta yet.

bubnikv, 2018-09-02:
> Published the release 1.41.0, but I don't see this feature :(

> We will implement automatic export of G-code in 1.41.
> Published the release 1.41.0, but I don't see this feature :(

Too bad, we did not manage yet.

timur-tabi, 2018-09-25:
I would be happy with a button that just said, "calculate print time" and did all the work of generating the g-code, but just didn't prompt me to save it anywhere.  This would be just a GUI change, then.  You can have it quietly print to /dev/null or something like that.

bubnikv, 2018-09-25:
This is currently being implemented in branches dev2 and dev_native.

jhoblitt, 2018-09-25:
I find that I am constantly exporting to `junk.gcode` just to get time estimates to see the impact of adjusting parameters.  Ideally, there would be an option to enable gcode to be generated in the background (after background slicing) and the generated gcode cached for export/send to printer.  That said, gcode generation seems to be generally < 10s, even with multiple parts layed out, so even a simple estimate build time button that throws away the produced gcode would be a big improvement over the save file dialog.

Patronics, 2018-09-25:
Yeah, I’ve adopted a similar technique, in my case saving gcode to a folder called “Junk drawer for time estimate” in the top level of my 3D printing folder, which automatically deletes anything saved into it (using an Automator folder action).

mkyral, 2018-10-16:
So when this is planned for release? Force save of gcode each time I change setting is annoying 🤨

And one more question. There is a field named Cost, but lack a unit. I have no idea what it is for.

trevjonez, 2018-10-16:
> And one more question. There is a field named Cost, but lack a unit. I have no idea what it is for.

In the filament settings you can set a cost for the material in use. "money/kg" it is just a constant value they use to calculate the material cost based on the usage determined by the job simulation.

mkyral, 2018-10-17:
Thanks, I expected something like this but I was blind. There are so many options…

bubnikv, 2018-12-11:
This will be part of the upcoming Slic3r 1.42 alpha release. Closing.

closed 2018-12-11 by bubnikv