prusa3d/PrusaSlicer #801: Add "reset to preset" button

grigorig, 2018-03-18:
### Version
1.39.1

### Behavior

When I play around with the settings, I often encounter the situation that I want to go back to a "clean slate", i.e. the current preset. It looks like this isn't easily possible right now, or at least not easily accessible. As a work around, I usually select a different preset and then switch back to the other preset again. But this is a bit annoying.

I think it might be the most convenient to place a third button in the preset selection row in the UI: after "save preset" and "delete preset", maybe add a "reset settings to preset" button? I would probably use a simple "undo" icon, which should be available in the standard GTK icon set.

bubnikv, 2018-03-19:
Interestingly enough, we are working on a richer feature than that. We are introducing user presets / system presets concept, and there will be an "undo" button for returning to a system preset value or your saved value, all nicely color coded. Stay tuned :-)

bubnikv, 2018-03-19:
This is something @YuSanka is working on at the front end and @vojtechkral at the backend.

bubnikv, 2018-04-18:
It will be part of the upcoming 1.40 beta, closing.

closed 2018-04-18 by bubnikv