prusa3d/PrusaSlicer #803: Feature Request: Guide Holes in object when cut

kwaad2, 2018-03-19:
Requested Feature: When you cut an object in two, allow the option to put "pins" that will cut out guide holes that are perfectly aligned to both sides of the cut part.
Pin could be simply a length of filament, or possibly even "user determined" diameter, length, and number of guide pins.

The problem
* After cutting an object it can be sometimes difficult to align them up perfectly when you are re-attaching them after printing. If you are using a non-instant set glue, you can be left holding the parts for some time for the glue to set.

Concept of "how to" involving the requested feature.
"Adjust the height of where you want to cut, Enable object transparency. Enter 1.75mm pin diameter, 3 alignment holes, with 5mm pin length. Drag the pins to where you want on the model (ensure the pin does not cut through the shell of the object). When you press "perform cut" you will receive both sides of the object, with your 3 alignment holes."

bubnikv, 2018-03-19:
I am afraid this is quite an advanced feature, which will not be implemented any time soon.

Why don't you use the MeshMixer or Fusion360 or any other CAD for this?

kwaad2, 2018-03-19:
I am not very good with Fusion 360, and I have yet to work with MeshMixer. (It's something that I plan on learning to do though)
I only posted it here (no hurry), as I thought it would be a good feature for an amateur with little to no experience with the advanced software. It would also be a bit of a stand-out feature (to me at least), as I don't know of anything that can do this in a "user friendly" method.

lukasmatena, 2023-02-01:
Implemented in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena