prusa3d/PrusaSlicer #805: Stops and extrudes, visualization errors

sindjinn, 2018-03-23:
### Version
1.39.1-prusa3d-win64
Win10

While printing the "heatbed-cable-cover.stl" the printer stops in the middle of hole and extrudes a lot of filament.   The amount of material reaches up to the top of the nozzle (possibly more, I stop it).  I believe this is an error in the slicing.

Things I have tried:
-  0.20 and 0.16 layer heights
-  different orientations for the parts
-  printed on different areas of the printbed

the .stl is the one that Prusa supplies for the 2.5 upgrade

the error shown in the visualization
[image]

the gcode with the error
[heatbed-cable-cover error.zip](https://github.com/prusa3d/Slic3r/files/1840064/heatbed-cable-cover.error.zip)

my print settings
[m 0.20 PETG.zip](https://github.com/prusa3d/Slic3r/files/1840065/m.0.20.PETG.zip)

FidelCapo, 2018-03-23:
@sindjinn Thanks for the report. I've tested your print settings and found the settings that caused it.

In print settings - Flow - Brigde flow ratio: You have set 90 (The correct setting is 0.8)

I suppose you wanted to enter 90%, but this setting does not allow you to enter percent.
In future print settings, we'll adjust this parameter so that the maximum value that can be entered is 2.

closed 2018-03-23 by FidelCapo

sindjinn, 2018-03-25:
Thank you so much for your help!!  Indeed, I thought that was a percentage.