prusa3d/PrusaSlicer #806: Under extrusion after layer change/travel/retraction

jonbet83, 2018-03-23:
### Version
1.39.1

I'm getting under extrusion after layer change/retraction/ travel move. nothing seems to resolve it.

Even with no retraction enabled the layer Z seam is there but straight after is slight under extrusion which leaves an imperfection visible to the eye compared to the layer above or below. Model run fine in S3D without this happening. 

Extra restart/de-retraction speeds didn't seem to have much of an effect, wondering if something more is going on...

slowing down perimeters to snail pace didn't have any good effects either.

Thanks

FidelCapo, 2018-03-26:
@jonbet83 I'm sorry, but I donť have enough data to try to reproduce your problem. Please send me STL and config files (File - Export Config) and what printer model do you have. Thank you

jonbet83, 2018-03-26:
Hi,

Thanks for the reply, I have some pictures and ill post the Gcode file.

[image]
[image]

Its a MK3 with latest firmware.

The problem isn't limited to that file its been happening with everything with slic3r so far.

jonbet83, 2018-03-26:
[GITHUB.zip](https://github.com/prusa3d/Slic3r/files/1847412/GITHUB.zip)

Heres the other files, sorry for the lack of detail in the first instance :)

FidelCapo, 2018-03-26:
@jonbet83 Can I ask about the kind of filament you used to print with this configuration? Because of the temperature in the configuration it does not look like PLA.

jonbet83, 2018-03-26:
Hopefully I’ve exported the correct one, it’s the first time I’ve exported a config, it was printed in Prusa PLA, 215 nozzle first layer with 210 after and a heated bed at 60. It was the material that was provided with the printer. 

Also if there’s any doubt with the files I sent, I was using the default profiles provided in slic3r PE for a mk3 afterwards with the exact same results. 

To rule out the filament I used some our decent quality and the results were the same. I was concerned it was “just me” or the printer it’s self so I done the same model in S3D and the problem wasn’t there.  

I drew the conclusion maybe it’s Slic3r. Most of the settings should be listed in the .gcode file. 

What have you noticed could be causing it?

jonbet83, 2018-03-26:
Apologies  for the mistake, please see the correct config export.

[MK3_config_new.ini.zip](https://github.com/prusa3d/Slic3r/files/1849838/MK3_config_new.ini.zip)

bubnikv, 2018-04-18:
We have investigated similar issue recently. It seems that our Slic3r profiles have the lower threshold of the "cooling slow down" set too aggressively, and reducing the print speed too aggressively leads to these issues. Will the print improve if you print multiple copies of the object at once? Printing multiple copies of the object at once will extend the time to print a single layer over the cool down threshold, so all external perimeters will be printed with the same speed.

jonbet83, 2018-04-18:
@bubnikv So i set up 9 of the cute little bunnys and let the print run, it did indeed fix the problem i was seeing. so thanks.

It does lead me to the question is the only way to stop this happening by disabling the min layer time or is there a fix coming?

i remember when i was looking at the gcode analyser the external perimiter was printed faster than the inside even with a slower external print speed set

bubnikv, 2018-04-25:
This topic is similar to https://github.com/prusa3d/Prusa-Firmware/issues/602
and we are currently reworking the cooling logic.

jonbet83, 2018-04-25:
I can see where the confusion is between this and some of the issues on 602 but this is a totally different problem (which your aware of the fix so it’s all good on this) 602 seems no further forward unfortunately

bubnikv, 2018-04-25:
@jonbet83 

> I can see where the confusion is between this and some of the issues on 602 but this is a totally different problem (which your aware of the fix so it’s all good on this) 602 seems no further forward unfortunately

I cannot make sense of it.

jonbet83, 2018-04-25:
I don’t think anyone can tbh, it must be as frustrating for us as it you at prusa, I know there’s loads of idea being bounded around, what’s your thoughts on it all?

bubnikv, 2018-05-10:
We hope that we have fixed it with the Slic3r 1.39.2 release with the updated cooling logic, see the change log of https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.2-beta

Would you please confirm that?

bubnikv, 2018-05-10:
> Yes I can confirm it was the hotend speeding up after a slower inner
> permitter, printing multiple items stopped it happening I can’t commnet on
> the newer slic3r I haven’t tried it yet, I can try it in a day or two if
> you want

yes, please test it with the 1.39.2 so we can close this issue. Thank you.