prusa3d/PrusaSlicer #807: Bridging regression on master (no bridge)

supermerill, 2018-03-25:
### Version : master
(ok on 39.1)
### Operating system type + version
windows 10 home

### Behavior
Seems like the bridge area is merged into the normal one. Anyway, it's not sliced correctly anymore.

#### STL/Config (.ZIP) where problem occurs
39.1 beta:
[image]

branch master:
[image]

[bridgeTest.zip](https://github.com/prusa3d/Slic3r/files/1845935/bridgeTest.zip)

FidelCapo, 2018-03-26:
Thanks for the report. I can not reproduce your issue. Can you please send your configuration file (File - Export config) so I can find the cause of the problem? Thank you

supermerill, 2018-03-26:
@FidelCapo i forgot to try an other config, as it's almost the standard one. It happens when you set the number of solid bottom layer to 0.

FidelCapo, 2018-03-28:
@supermerill This is an unusual combination of settings and object. You have 4 solid top layers, 0 solid bottom layers and 20% infill. Slicer does not have to do (and do not do) bridges at all because it has no bottom solid layers. What you see in the first layer as "bridge" is the first top solid layer. If you set the top solid layers to 2, you will see that this first layer "bridge" will only be filled in by the infill value.

bubnikv, 2018-04-18:
@supermerill What should be the expected behavior? Do you expect Slic3r to produce bridges even if you have zero bottom layers configured?

supermerill, 2018-04-18:
@bubnikv yes. I see no adverse effect.
I was messing a bit with bottom /bridge layer on my side, and discover that. It's detected as bridge because it's over nothing but somehow, it doesn't translate to a bridge behavior.
If you don't have the time to mess with it, i can look a bit more into that. Depends if you agree with it or not.

bubnikv, 2018-05-10:
> If you don't have the time to mess with it, i can look a bit more into that. Depends if you agree with it or not.

A helping hand is always helpful :-) A single line fix is even more helpful.

bubnikv, 2019-09-06:
The issue "happens" here when renaming the surface.is_bottom() to stInternal.
Just replacing surface.is_bottom() with a test for stBottom will work in your case, but it will break the expectation of somebody who want to print just the bridging perimeters without the bottom bridge.

```
void LayerRegion::prepare_fill_surfaces()
{
    if (this->region()->config().bottom_solid_layers == 0) {
        for (Surface &surface : this->fill_surfaces.surfaces)
            if (surface.surface_type == stBottom) // (surface.is_bottom())
                surface.surface_type = stInternal;
    }
```

If this change was accepted, then one would likely want to convert the bottom most top layer to a 
bridging layer, if it looks now like a bridge. These are corner cases and I am not sure whether they are worth the additional code complexity. What do you think?

closed 2026-09-29 by prusaslicer-bot[bot]