prusa3d/PrusaSlicer #808: [Bug] Overhang Threshold Adjustment Crash

sarf2k4, 2018-03-26:
### Version
1.39.1

### Operating system type + version
Win 10 64bit

### Behavior
Adjusting the Support>Overhang Threshold eventually crash the slicer if the value is too low. I've tested several times with part 3 of the model below ranging between 10-15' in an increment of 1', the slic3r kept crashing and had to reload the slic3r.

Same goes for part 2, changed to 4' from the default 45', slicer just crashed. Meant to put in 44'.

This only occurs when "background processing" was turned on.
#### STL/Config (.ZIP) where problem occurs
[Link to the file is here](http://www.thingiverse.com/thing:391664)

bubnikv, 2018-03-26:
It may really be connected to the background processing. I just reviewed
how the background processing is implemented and there are ugly race
conditions. I am currently rewriting the background processing part.

Are you able to reproduce the crash with background processing disabled?

sarf2k4, 2018-03-26:
@bubnikv, I'm unable to produce the same crash/problem if I were to turn off "background processing". Even with the same value I inserted to produce the crash when "background processing" were turned on, and same model too

bubnikv, 2018-03-26:
Thanks. I will concentrate my efforts to fix the background processing
then. There are quite some number of issues connected with the background
processing and the background processing cancellation (which happens when
you delete an object from the plater and the calculation is running for
example).

sarf2k4, 2018-03-26:
I'm no advanced programmer, but maybe you could add a bit of delay before the background processing starts to do its job after a parameter has been changed probably around 1 second or adjustable delay perhaps?

bubnikv, 2018-12-11:
The code responsible for your issue was rewritten from Perl to C++, also the background processing should now work reliably. Please retest with 1.42 alpha once it is released this or the next week. Closing.

closed 2018-12-11 by bubnikv