prusa3d/PrusaSlicer #810: Sequential printing not working

ir-fuel, 2018-03-27:
### Version

1.39.1-beta1

### Operating system type + version
macOS 10.12.6

### Behavior

I tested it in 2 different ways. One was duplicating an imported object, second one was importing 2 stl files.

Print settings >> Output options >> "Complete individual objects" is checked, yet after slicing in the `Preview` window I can see that both objects are printed as if the checkbox was not checked. It just prints both of them layer by layer.

#### STL/Config (.ZIP) where problem occurs

Any object I have tried.

[image]
[image]

bubnikv, 2018-03-27:
The G-code is correct, just the preview does not display the lines by time, but by layers.

dartrax, 2018-04-03:
I had the same effect today, assuming that it was a fault of slic3r while indeed it generates GCode correctly. Second, gcode.ws does not show a sequential preview also, even if "sort by Layer Z" is unticked.
So, while a sequential gcode preview would be very nice (S3D has it), a hint that is not visible in preview could be added to the tooltip in the meantime.
Is there a way to check in what order the objects get printed? This is important to avoid collisions.
The only way I found was to Delete the objects one by one, the first that is removed ist the last that gets printed, and so on.

bubnikv, 2018-04-03:
> So, while a sequential gcode preview would be very nice (S3D has it), a
hint that is not visible in preview could be added to the tooltip in the
meantime.

Which tooltip? The "sequential print"?

> Is there a way to check in what order the objects get printed? This is
important to avoid collisions.

No, there is currently no way. They are printed in the order they are added
to the platter.

I believe the mainline Slic3r has a way to visualize and change the order,
we may implement something similar in 1.41.

dartrax, 2018-04-03:
The "Complete individual objects:"-Tooltip.

bubnikv, 2020-02-04:
The upcoming PrusaSlicer 2.2.0-alpha4 will print sequentially in the order displayed at the object list. Also the object list will support drag & drop to reorder.
Closing.

closed 2020-02-04 by bubnikv