prusa3d/PrusaSlicer #811: Export G-code takes a very long time - unusual behavior

vernieg, 2018-03-28:
### Version
1.39.1-prusa3d

### Operating system type + version
macOS HighSierra 10.13.3 (17D102)

### Behavior
* _Describe the problem_
Consistently takes a very long time to generate the gcode on some models with layer editing done on some layers. 

* _Expected Results_
It should only take seconds, it doesn't even take a minute for most.

* _Actual Results_
Several minutes, feels like forever but it will generate.
There is one instance only where it totally failed, saying the tmp file could be lock?

#### STL/Config (.ZIP) where problem occurs

[stl file and config.zip](https://github.com/prusa3d/Slic3r/files/1854154/stl.file.and.config.zip)

bubnikv, 2018-04-18:
@FidelCapo Would you please verify this problem? Thanks.

bubnikv, 2018-05-10:
@vernieg Don't you have the "avoid external perimeters" feature enabled?

bubnikv, 2018-05-11:
> I can't recall that setting. What I see here is "Avoid crossing
> perimeters" enabled.

will the export take reasonable time if you disable this feature? do you
need this feature? The avoid external perimeters feature is problematic,
see the issues marked with the "path planning" tag.

vernieg, 2018-05-11:
The export is fast (almost instant) with this feature disabled. Thank you.  The first time I saw the option group about quality, I simply enabled them hoping they do contribute better quality.  When it comes to quality, I never went back to using another slicer, slic3r prusa edition is where I am happy with the results specially when dealing with tiny details.