prusa3d/PrusaSlicer #813: Missing middle layers on Preview - 1.39.1-prusa3d-linux64

vernieg, 2018-03-29:
### Version
1.39.1-prusa3d-linux64

### Operating system type + version
Linux Mint 18.3 Sylvia, 4.16.0-999-lowlatency #201803282203 SMP PREEMPT Thu Mar 29 02:09:31 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
Preview with missing middle layers.

[image]

#### STL/Config (.ZIP) where problem occurs
[config-and-stl-file.zip](https://github.com/prusa3d/Slic3r/files/1860096/config-and-stl-file.zip)

enricoturri1966, 2018-03-29:
I have tried this model with our latest alpha and it gets sliced properly by it.

bubnikv, 2018-03-29:
@FidelCapo Please add this model to our test set.

EvilGTV, 2018-04-11:
I've just figured something strange out - I've been having trouble with slicing a couple of things, most annoyingly the cover for the Drakon extruder where it just wanted to fill in the dragon logo.  Also I had the same problem slicing Benchy inside Repetier as someone else here, and then I came across this dice file here while looking and I had the same problem as the OP.

But this is where things get.... absolutely bizzare.  Hopefully bizarre enough to be rather helpful tracking down the source of a few faults.  Looking through the previous version update history, I saw mentioned the "--gui" switch.  So I thought I'd try adding it after the executable path inside Repetiers slicer manager, to see what happens.

So now, the dice in this post slices absolutely perfectly in Repetier.  Excellent news, you say.  The bizarre part is that the dice still doesn't slice properly if I open SlicerPE directly, even if I use the same EXE that Repetier is using.

Same thing with the Drakon extruder cover (found here: https://github.com/AxMod3DPrint/Drakon-Extruder/blob/master/Drakon_Cover.stl).  Now finally slices just fine in Repetier host, but not in SlicerPE natively.

Benchy already sliced fine in SlicerPE natively, but now also works properly in Repetier host...

I'm using the same version of SlicerPE as the OP, but on Windows 8.1.

bubnikv, 2018-05-10:
This will be fixed in Slic3r 1.40.0-alpha. Closing.

closed 2018-05-10 by bubnikv