prusa3d/PrusaSlicer #815: [BUG] 200% extrusion width is applicated only on first object on buildplate

pinkpiggy, 2018-03-29:
### Version
Slic3r 1.37.2 - Prusa (maybe the newer ones as well, didnt try)

### OS
Hackintosh (Mac OSX El Capitan)

### Printer
Rebelix X2

### Behavior
2 simple objects on single buildplate (XY_Idler from Hypercube) in a same time printed different width of first layer. My own settings for first layer extrusion width is 200% of first layer height 130% of layer height 0.2mm. It is not visible from the slic3r preview but its pretty recognizable from the print. While first object is nicely printed with this settings, second one use "basic" settings of every other layer (0.2mm) and it doesnt count with that 200%.
Adding GCODE
[XY_Idler_2x.gcode.zip](https://github.com/prusa3d/Slic3r/files/1862201/XY_Idler_2x.gcode.zip)

dartrax, 2018-04-03:
Sure that it is a problem with GCode? The generated GCode of first layer left and right are looking identical (same color, color = extrusion mm per move mm) in gcode.ws 
Is the distance between nozzle and build plate equal?

[image]

pinkpiggy, 2018-04-03:
@Dartrax
the distance between nozzle and build plate is pretty equal. I did measurement before print and everything was ok. As well as when i print 1 bigger object (or even smaller on the position of the second object), everything is fine. Only possibility i see is Slic3r OR repetier firmware (i am using 0.91)

bubnikv, 2018-05-10:
@FidelCapo Would you please test it? Thanks.

FidelCapo, 2018-05-15:
Can you please provide me a STL and config files? (File - Export config...) Thank you

bubnikv, 2018-05-16:
I am sorry, but I do not understand.

Please provide as much data as you can. Provide the STL, config file and a G-code. If you printed multiple objects, it is best to export the AMF file as well.