prusa3d/PrusaSlicer #816: Errors and missing layers on preview when slicing from Repetier Host

Itox001, 2018-03-31:
### Version
Slic3r PE 1.39.1

### Operating system type + version
Windows 7 64 bit

### Behavior
When slicing Benchy (yes, the good old benchy) I get missing layers in the Repetier Host 2.0.5 preview, and lots of messages saying "Unable to close this loop having X points" in the console. It looks like this: 
[image]

If I slice the same good old Benchy directly on Slic3rs plater, the preview seems OK, but I also get a message saying that the model had 1104 errors and was autorepaired. It looks like this: 
[image]

Any idea on why this happens? When using Slic3r PE 1.37.2 (the one I had before upgrading) it doesn't happen and the preview looks OK even when sliced from Repetier Host and doesn't throw any errors. Also, if I export the Gcode to a file directly from Slic3r 1.39.1 and then import it in Repetier, it looks ok, so it seems like the problem is when slicing directly from Repetier. Using the no console exe.

#### STL/Config (.ZIP) where problem occurs
[ConfigSTL.zip](https://github.com/prusa3d/Slic3r/files/1865171/ConfigSTL.zip)

EvilGTV, 2018-04-11:
Try this:  Go into the slicer manager in Repetier Host, and add to the end of your path to Slic3rPE executable: --gui  

so you should have:

C:\whatever directory\Slic3r-1.39.1-prusa3d-win64-full-201803010855\slic3r-console.exe --gui

Doesn't matter if you use console or noconsole, worked on both for me.

If it works, perhaps you could post a note here, where I've replied to another post:
https://github.com/prusa3d/Slic3r/issues/813

I'm hoping this might be the key to unlocking a few issues, if you read my post there you'll see I believe there are a couple of connected bugs, this being one of them...  If we can help tie them together, it should help to narrow it down!

BerndiDasBrot, 2018-04-13:
I had the same problem. It worked for me.

Itox001, 2018-04-17:
When I do that, Repetier reverts to using the default Slic3r PE 1.36.2 for some reason. If I remove the --gui option, then it uses 1.39.1. Weird. Any ideas?

EvilGTV, 2018-04-17:
You're right - I hadn't noticed that, since I hadn't actually specifically checked the version number in the config window inside Repetier.  So setting --gui somehow makes it completely ignore the path you've set, which seems rather stupid.

And no, I have no idea!

repetier, 2018-04-23:
I made some tests as I'm currently updating slicers in Repetier-Host. The problem is not the --gui parameter, it is handling of the model. When you slice over host model gets exported as uncompressed amf file, which I also attached here, and that model gets the errors.

It gets also the same errors if I add the amf file directly in slic3r, but if I use the original benchy it works. On the other side if I slice it with latest original slic3r from 18.4.2018 it gets sliced with no errors.

I also saw that other issues also have missing layers. Maybe it is a related thing. Did you change any tolerances, so interpretation now causes different mesh? Also amf has already removed double nodes it might happen in a later step that points with close distances now get merged causing invalid meshes.

[composition.zip](https://github.com/prusa3d/Slic3r/files/1938625/composition.zip)

Itox001, 2018-04-29:
Tested the new 1.39.2 beta, problem still persists.

Itox001, 2018-05-10:
@bubnikv where can I find the alpha? I'd like to try it.

bubnikv, 2018-05-16:
Would you please test with the 1.40.0-alpha1?
https://github.com/prusa3d/Slic3r/releases/tag/version_1.40.0-alpha1

We improved handling of buggy meshes, and I can slice the amf file attached to this issue without any problem.

Itox001, 2018-05-16:
Tried it yesterday, it's now working properly. Good job on the GCode visualizer, it's awesome. Issue closed.

closed 2018-05-16 by Itox001