prusa3d/PrusaSlicer #817: It seems the multi-color is limited to 4 extruders, and the smart wipe tower doesn't work. can you remove it or increase it to 10 or more? thanks.

cnqdlcq, 2018-03-31:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-04-18:
You should be able to enter the number manually using the keyboard. Does it work for you?

ichbinscheisse, 2018-05-03:
Yes, entering the number by keyboard works and wipe tower includes those color.

With a certain amount of colors the wipe tower will exceed the print bed dimensions.
Could the wipe tower start a new row as soon as the end of the print bed is reached?

lukasmatena, 2018-05-10:
Hi. The wipe tower will be more customizable in the upcoming release of Slic3r, which will be out in two weeks time at the latest. It will allow you to spread the wipe tower across the entire print area, if you so choose. Please, test the feature after the release, provide feedback and close the issue if happy. Thank you.

YuSanka, 2018-09-05:
Correct extruder selection is implemented here (https://github.com/prusa3d/Slic3r/commit/3e549c153d508ba2cd5cea41ee8bd35b53b87596) and will be a part of Slic3rPE 1.42

bubnikv, 2018-12-11:
Slic3r PE 1.42 alpha is to be released soon. Closing.

closed 2018-12-11 by bubnikv