prusa3d/PrusaSlicer #818: Quality of life: Custom gcode editor

dtandersen, 2018-03-31:
* CTRL-A should allow select all in custom gcode text boxes
* Tool tip should be moved from the text box to the label so it doesn't cover text box

Panayiotis-git, 2018-04-02:
The tooltip could also be moved to an (i) icon next to the field.
Also, if there are a lot of lines in the text box, it shouldn't scroll to the top when we switch to another window and then return back.

bubnikv, 2018-05-10:
I know @YuSanka worked on it. What is the status?

YuSanka, 2018-05-11:
Hi, all! Current status is this:
* Ctrl-A works and is in master branch now 
* Custom Gcode text boxes don't have labels, so at this time it works this way: if you click inside of the text box, tooltip will disabled.

bubnikv, 2018-05-11:
It should work in the Slic3r 1.40 alpha, which will be released in couple of days. Closing.

closed 2018-05-11 by bubnikv