prusa3d/PrusaSlicer #819: [Feature request] Support material: generate on print only

manuel-binpar, 2018-03-31:
Hello folks

As it sounds, I would like to know if you can consider to build an option to select where the support material will be printed on. I'm using a lot of TPU and support material generated on the build plate is not useful in certain cases.

Thank you and keep the good work!

bubnikv, 2018-05-10:
Do you know any slicer that supports such an option? Would you please be more specific as why you don't want to print supports on the print bed? Maybe some pictures will help.

bubnikv, 2018-08-02:
Sorry, this is such an exceptional request, that it will only clutter the user interface to vast majority of users. We will not implement such request.

closed 2018-08-02 by bubnikv