prusa3d/PrusaSlicer #820: Bug: Tiny gap gets ignored and replaced with infill.

JoschaMiddendorf, 2018-04-01:
### Version
1.39.1-prusa3d

### Operating system type + version
macOS High Sierra 10.13.4 (17E199)

### Behavior
* _Describtion the problem_
  * _Slic3r is just generating infill where a gap should be._
* _Steps needed to reproduce the problem_
  * _slice this [part](https://gumroad.com/l/OxKwn)_
* _Expected Results_
[image]
* _Actual Results_
[image]

#### STL/Config (.ZIP) where problem occurs
[ConfigAndSTL.zip](https://github.com/prusa3d/Slic3r/files/1866737/ConfigAndSTL.zip)

tkircher, 2018-04-12:
It's not really a bug. Features below a certain size are ignored on purpose.

bubnikv, 2018-05-10:
No, it is an issue with the Clipper library when handling polygons with holes. The issue is similar to
https://github.com/prusa3d/Slic3r/issues/865
https://github.com/prusa3d/Slic3r/issues/676

bubnikv, 2018-05-16:
@tkircher was right. It is the same issue as https://github.com/prusa3d/Slic3r/issues/520, slic3r closes gaps smaller than 0.01mm on purpose, as they are not really printable by the FDM technology, or at least they were not some years ago. We may either lower the threshold, or just remove it, but not for the 1.40 release.

To verify the hypothesis, lower the safety_offset in TriangleMesh.cpp, just search for
//FIXME see https://github.com/prusa3d/Slic3r/issues/520
    double safety_offset = scale_(0.0001);

I am closing this issue as a duplicate of #520, this issue will be referenced from #520.

closed 2018-05-16 by bubnikv

bubnikv, 2019-03-01:
By the way, I have introduced a new parameter "slice_closing_radius", which is set to 0.049 by default, leading to the same results as in 1.41.3. If you need to maintain the thin gaps, starting with Slic3r-1.42.0-alpha7 you will need to lower the new "slice_closing_radius" value.

Implemented with e3cba0a65a691b77be87f8440c91ec5f4916c3b8

This parameter and the return to the 1.41.3 behavior by default was made as we have discovered some models with cracks, which were handled correctly by 1.41.3, but which were full of spurious perimeters in 1.42.0-alpha5. We find it safer to revert to the 1.41.3 behavior while giving you the parameter to tinker with.