prusa3d/PrusaSlicer #824: Gyroid not filling this model correctly

foreachthing, 2018-04-05:
### Version
1.3 dev Slic3r (_not_ Prusa!)

### Operating system type + version
Windows 7 x64

### Behavior
(According to @supermerill I am to report this issue here)

- Load part attached
- Load config
- Switch to Layers tab and observe infill: 
[image]
  - Infill missing in red area

Expected: "same" infill as Cubic:
[image]

#### STL/Config (.ZIP) where problem occurs
[Gyroid_Issue.zip](https://github.com/prusa3d/Slic3r/files/1878626/Gyroid_Issue.zip)

vojtechkral, 2018-04-05:
Yes, we've observed the issue as well. I believe it is fixed by 4611b5094e87c93faf95d36a2690ec232dc5e962 in Slic3r PE master.

closed 2018-04-05 by vojtechkral

foreachthing, 2018-04-05:
Ok ... thanks!
Now: how can I get this to slic3r/Slic3r?
Or, how can I get a working Windows version of Slic3rPE with this fix? I'm not allowed to install all the required software to compile it myself...

bubnikv, 2018-04-05:
It is just the line 77 of
https://github.com/prusa3d/Slic3r/commit/4611b5094e87c93faf95d36a2690ec232dc5e962
Easy fix, single word typo.