prusa3d/PrusaSlicer #825: Gyroid does not respecting Overlap setting

foreachthing, 2018-04-05:
### Version
1.3 dev Slic3r (_not_ Prusa!)

### Operating system type + version
Windows 7 x64

### Behavior
(According to @supermerill I am to report this issue here)

- Load part attached
- Load config
- Switch to Layers tab and observe infill OVERLAP.

Expected: "same" infill and overlap settings as Cubic:
[image]

Zoom image. Blue: Gyroid vs. black: Cubic.
[image]

Note that Gyroid does not honor **Overlap** (Print Settings -> Advanced -> Overlap: Infill/perimeter overlap) setting.

#### STL/Config (.ZIP) where problem occurs
[Gyroid_Issue.zip](https://github.com/prusa3d/Slic3r/files/1878626/Gyroid_Issue.zip)

bubnikv, 2018-05-19:
@foreachthing @supermerill I am confused. Shall I worry about the Slic3r PE? Do we have it or is the bug just in the upstream Slic3r? Thanks for taking time and helping with the investigation.

foreachthing, 2018-05-20:
Uhm ... no! You don't have to worry about Slic3rPE! I just tested this version: Slic3rPE-1.40.0-alpha1+win64-full-201805141300 and it does what it should. It's only in Slic3r (<= Slic3r-master.2018.05.16) where it's not working.

bubnikv, 2018-05-22:
@foreachthing Thanks for the report. Then please report the issue at https://github.com/slic3r/Slic3r/issues

Closing.

closed 2018-05-22 by bubnikv

foreachthing, 2018-05-22:
Giving me the run around, arentcha?! :-)
I came here, beacause @supermerill  said so. Oh, well! I am off to see the wizard on the other side of the forest.

supermerill, 2018-05-22:
@foreachthing as you say "According to @supermerill I am to report this issue here", it seems I misinterpreted something. Where did i say that to you? I don't remember. Sorry everyone for the inconvenience if I made a mistake.

foreachthing, 2018-05-23:
@supermerill: I misread your answer in https://github.com/slic3r/Slic3r/issues/3666. Of course it's not Slic3rPE related (which I didn't know for sure at this time). My sincerest apologies!