prusa3d/PrusaSlicer #827: compiling in OSX

v-mat, 2018-04-05:
### Version
1.38.5 and 1.39.1

### Operating system type + version
OSX 10.12.6

### Behavior
Have been trying to compile slic3d with no luck. Following instructions on:

https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X

Getting stuck at "BOOST_DIR=~/boost_1_64_0 SLIC3R_STATIC=1 perl Build.PL".  This seems to run fine, but when I invoke "perl slic3d.pl", I'm getting this message:

"Can't locate Slic3r/XS.pm in @INC (you may need to install the Slic3r::XS module) (@INC contains: ..."

I saw the updated instruction "How to build - Windows.md" under /doc of the 1.39.1 (mentioned in #781), but not for OSX.. 

Any help would be appreciated.

vojtechkral, 2018-04-05:
That wiki page is likely to be out of date. I'll have a look and try to help...

v-mat, 2018-04-05:
Thanks for the quick reply. I've kept digging, but haven't made much progress. I've also tried to compile with cmake (by running "cmake .. -DCMAKE_BUILD_TYPE=Release"),  but this ends with "-- Configuring incomplete, errors occurred!".  Here are some the error messages shown:

"Imported targets and dependency information not available for Boost version"
"Unable to find the requested Boost libraries."
"Could NOT find TBB (missing: TBB_INCLUDE_DIRS TBB_LIBRARIES)"

v-mat, 2018-04-05:
Thanks bubnikv. I'm afraid "export SLIC3R_STATIC=1" didn't help with the BOOST issue, but "cmake .. -DBOOST_ROOT:PATHNAME=~/boost_1_64_0/" seems to have taken care of it.

Now I'm running into the following:

"Could NOT find TBB (missing: TBB_LIBRARIES) (found version "2017.0")"

I actually don't see a /lib under my /tbb, nor in https://github.com/wjakob/tbb...

bubnikv, 2018-04-05:
>  I actually don't see a /lib under my /tbb, nor in
https://github.com/wjakob/tbb...

You have to build tbb first.

v-mat, 2018-04-05:
Oops, you're right. I've just built it with brew (couldn't find how to with cmake) and set the environment variables as you said above. I'm still running into the same error though, i.e.:

"Could NOT find TBB (missing: TBB_LIBRARIES) (found version "2018.0")"

If I do have to compile with cmake, could you please share how?

bubnikv, 2018-04-06:
To build tbb statically, just

unpack https://github.com/wjakob/tbb
go to the tbb directory
create a build directory
change to the build directory
run
cmake .. -DTBB_BUILD_SHARED=OFF -DTBB_BUILD_TESTS=OFF
-DCMAKE_POSITION_INDEPENDENT_CODE=ON

That should be all to it.

bubnikv, 2018-05-10:
Any progress on your compilation efforts?

v-mat, 2018-05-10:
Thanks for asking. I did managed to build tbb as per your previous post and I can run cmake without any errors, but I still get the same original error message when invoking "perl slic3d.pl" (full error message below):

<<<
Can't load '~/slic3r/Slic3r/local-lib/lib/perl5/darwin-thread-multi-2level/auto/Slic3r/XS/XS.bundle' for module Slic3r::XS: dlopen(~/slic3r/Slic3r/local-lib/lib/perl5/darwin-thread-multi-2level/auto/Slic3r/XS/XS.bundle, 1): Symbol not found: __ZN3tbb4task13note_affinityEt
  Referenced from: ~/slic3r/Slic3r/local-lib/lib/perl5/darwin-thread-multi-2level/auto/Slic3r/XS/XS.bundle
  Expected in: flat namespace  
 in ~/slic3r/Slic3r/local-lib/lib/perl5/darwin-thread-multi-2level/auto/Slic3r/XS/XS.bundle at ~/perl5/perlbrew/perls/threaded-perl-blead/lib/5.27.11/XSLoader.pm line 96.
 at ~/slic3r/Slic3r/local-lib/lib/perl5/darwin-thread-multi-2level/Slic3r/XS.pm line 22.
Compilation failed in require at ~/slic3r/Slic3r/lib/Slic3r.pm line 50.
BEGIN failed--compilation aborted at ~/slic3r/Slic3r/lib/Slic3r.pm line 50.
Compilation failed in require at slic3r.pl line 16.
BEGIN failed--compilation aborted at slic3r.pl line 16.
>>>

bubnikv, 2018-05-14:
What will 
otool -L ~/slic3r/Slic3r/local-lib/lib/perl5/darwin-thread-multi-2level/auto/Slic3r/XS/XS.bundle
print out?

It seems like you are still linking tbb dynamically, so there is inconsistent linkage for some reason.

v-mat, 2018-05-14:
otool -L ~/slic3r/Slic3r/local-lib/lib/perl5/darwin-thread-multi-2level/auto/Slic3r/XS/XS.bundle prints this:
<<<<
local-lib/lib/perl5/darwin-thread-multi-2level/auto/Slic3r/XS/XS.bundle:
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1349.92.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.5.0)
	/usr/local/opt/wxmac/lib/libwx_baseu-3.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	/usr/local/opt/wxmac/lib/libwx_osx_cocoau_core-3.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	/usr/local/opt/wxmac/lib/libwx_osx_cocoau_adv-3.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	/usr/lib/libcurl.4.dylib (compatibility version 7.0.0, current version 9.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.60.2)
>>>>

Any change you could provide me with a TBB binary dist pre-compiled for static linking? My OS is OSX 10.12.

bubnikv, 2018-05-15:
Unfortunately that gave me no hint. It looks that this XS.bundle was
compiled with static TBB.

> Any change you could provide me with a TBB binary dist pre-compiled for
static linking? My OS is OSX 10.12.

I don't think that is the problem. More likely you have mixed up multiple
versions of Slic3r or TBB on your system.

v-mat, 2018-05-15:
> I don't think that is the problem. More likely you have mixed up multiple
> versions of Slic3r or TBB on your system.

I followed the given instructions, although they were somewhat fragmented. I do appreciate you trying to help. I look forward to the updated wiki for OSX.

ctgreybeard, 2018-06-29:
I, too, am trying to build on MacOS. I got boost resolved by building it locally but I cannot get TBB to be recognized at all.

I cloned tbb from github and built it which seemed to go OK. I do get the three libtbb _static.a files built.

I get
```
CMake Error at /sw/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find TBB (missing: TBB_LIBRARIES) (found version "4.3")
Call Stack (most recent call first):
  /sw/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindTBB.cmake:273 (find_package_handle_standard_args)
  xs/CMakeLists.txt:562 (find_package)
```

I have set the environment variables correctly I think: `TBB_LIBRARY=/Volumes/Ext1/Projects/tbb/build` which is where the libtbb files got created.

I could just wait for Monday when the RC will get updated but I have a project I'd like to do over the weekend.

bubnikv, 2019-01-07:
@vojtechkral is polishing a build description for 1.42 PE. Closing this old issue discussing the Perl based build system.

commit 00c9d1f1 (2019-01-07): doc: Add build tutorial for Mac
Fix #1421
Fix #827
Fix #1039
closed 2019-01-07 by bubnikv