prusa3d/PrusaSlicer #828: Feature request: Add "Hybrid Sequential Printing" for when some objects are too tall

Patronics, 2018-04-05:
### Version
Slic3r PE 1.39.1 (not really relevant)

### Operating system type + version
_What OS are you using, and state any version #s_
MacOS (not really relevant)
### Behavior
* _Describe the problem_
When slicing multiple objects where some (but not all) are taller than the extruder clearance, slic3r should offer to slice them such that it still prints as many of the short objects as possible sequentially, then moves on to print the taller ones simultaneously.
* _Steps needed to reproduce the problem_
 try to slice any 3 or more objects where some are taller than the extruder clearance.
* _Expected Results_
Slic3r should automatically (at least optionally) print all the short objects it can sequentially, then go print all the "too tall" objects simultaneously.
* _Actual Results_
When slicing multiple objects of varying height with "Complete Individual objects", Slic3r refuses to slice it if any 2 objects are too tall to print sequentially.

_Is this a new feature request?_
Yes.
#### STL/Config (.ZIP) where problem occurs
Any 3 or more objects where 2 or more of them are above the extruder clearance height

Also (not sure if this should be it's own issue), but is the default extruder clearance in Slic3r PE settings the correct clearance for the Prusa Printers? If not, it should probably be adjusted to default to the correct value.

scalablecory, 2018-04-05:
It seems a better feature would be more dynamic: print as much of an object as clearance would allow before moving to the next one. This way all the small ones would be printed perfectly, while the large one would print some millimeters at a time.

Patronics, 2018-04-06:
@scalablecory that would be interesting! My intuition says that might risk creating seams when it transitioned to another object for a while, before going back to the now much cooler object. I don't have a ton of experience with 3D printing yet though, so that intuition may very well be wrong, in which case that sounds like a good technique.

onionhammer, 2026-09-02:
Anyone know if this is fixed in 3.0?

Jan-Soustruznik, 2026-10-01:
This issue/behaviour should be fixed in the 2.9.1-alpha1 release with the implementation of the Arrange feature for sequential printing. I am closing this issue as resolved.

closed 2026-10-01 by Jan-Soustruznik