prusa3d/PrusaSlicer #829: Feature Request: More/other buttons for rotations

yschroeder, 2018-04-06:
More often than not, I design my own models. The STL export of my CAD program always exports the way I designed my model and this often does not match the best print orientation.

Then, I have to rotate my models 180° around X (or Y) because the are upside down (sometimes only 90°). There are very few cases where I need to rotate my model in 45° steps around Z and yet these are the only buttons the GUI offers.

I end up most of the time in the context menu rotating via the dialog box and hacking in the values by hand.

Is this only me or do other people experience this as well?
Maybe I am the only one with this problem, maybe not. If others have the same problem, please add your comment (this might be more of a discussion).

So my request would be: Have more (maybe smaller) buttons for these common rotations around the different axes.

bubnikv, 2018-04-06:
@enricoturri1966 will rework the 3D scene manipulators in the Slic3r PE 1.41.
@YuSanka will provide a UI panel for manipulating the object transformation matrix in a similar fashion to PrusaControl.

leobel96, 2018-07-10:
In my opinion it would be fantastic the possibility of choosing which UI buttons and where should be shown, not only to add rotation buttons. An ability to completely customize UI buttons

bubnikv, 2018-12-11:
We have implemented rotation gizmos, place on face and rotation by X/Y/Z from the platter's right panel. I believe this should fit your request. The changes will be part of the upcoming Slic3r 1.42 alpha release. Closing.

closed 2018-12-11 by bubnikv