prusa3d/PrusaSlicer #830: *BUG REPORT* Keyboard shortcuts interfering with program initialization

ctilley83, 2018-04-06:
### Version
1.39.1-prusa3d

### Operating system type + version
Mac OS X 10.13.4

### Behavior
Upon a clean install of slic3r PE, if you have any keyboard shortcuts on your system, it does not ignore them. So when slic3r is initializing the "Custom G-Code" portion of the software, if you have a keyboard shortcut "S0" or "T0" it populates the text you have assigned to those shortcuts.

[Here is a screenshot](http://hermeswebservices.com/slic3r_bug.jpeg)

_Is this a new feature request?_
BUG REPORT

bubnikv, 2018-04-18:
Unfortunately this is a feature of the wxWidgets UI toolkit, which Slic3r uses.
We would have to patch the wxWidgets library on OSX to disable the macro expansion. Actually we did that already once for the "beautification" of punctuation characters, but to do it once again requires quite some effort.

So one day we will fix that, but I am not sure whether we will manage it in 1.40.

bubnikv, 2019-01-28:
It will be fixed in the upcoming Slic3r 1.42.0-alpha4
Closing

closed 2019-01-28 by bubnikv