prusa3d/PrusaSlicer #833: [REQUEST] boot and idle fan speed

BETLOG, 2018-04-07:
Most fans have a minimum PWM required before they will begin spinning from no rotation.
However if a fan is already spinning it will continue to do so with extremely low PWM frequency. Usually much lower than required to motivate it from a stationary condition.

Adding the options below would enable the user to use significantly lower fan speeds.

* "boot" duration (ms)
Number of milliseconds the fan is driven at 100% PWM to get it spinning enough for idle speed to engage.

* "idle fan speed" (%)
PWM % required to keep an already moving fan barely spinning.

* off or idle when not in use (pulldown option menu - off/idle)
After any operations above the existing "fan speed min" value complete, and the fan would normally be turned off, this option would allow the fan to idle at minimal speed instead.

bubnikv, 2018-05-10:
We had this discussion before, we believe it is better to let the firmware do it. At least our Marlin branch has the fan kick-in enabled and it is a standard Marlin feature. What firmware are you running?

BETLOG, 2018-05-11:
Re-reading what I wrote just now I was noticing that what i mean may not be obvious enough. The last bullet point is really the main point.
ie: option to remain spinning very slowly as an alternative to 'off', as well as an actual 'off' setting.

Marlin 1.1.8 and bugfix2

closed 2026-09-29 by prusaslicer-bot[bot]