prusa3d/PrusaSlicer #834: Crashes (Segmentation Fault?) when slicing large, complex STL

Patronics, 2018-04-09:
### Version
1.39.1

### Operating system type + version
MacOS 10.13.3

### Behavior
* _Describe the problem_
Slic3r crashes when slicing this detailed globe file (included in the zip).
* _Steps needed to reproduce the problem_
import globetest1_north.stl into slic3r, clicking OK when it offers to scale it down to fit the bed. Click slice.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
it slices the file successfully.
* _Actual Results_
Slic3r immediately crashes, and user is presented with the crash log (included in zip as crashReport.txt).  The report seems to indicate the crash is caused by a segmentation fault.

_Is this a new feature request?_
No
#### STL/Config (.ZIP) where problem occurs
[Slic3rCrashData.zip](https://github.com/prusa3d/Slic3r/files/1891216/Slic3rCrashData.zip)

Patronics, 2018-04-09:
Additional experimentation reveals that with this remeshed to 90,000 triangles (with meshmixer) version of the file [globetest1_north_reduced_tris_90k.stl.zip](https://github.com/prusa3d/Slic3r/files/1891290/globetest1_north_reduced_tris_90k.stl.zip) It will successfully slice the file at the default scale it reduces it to after importing, but will immediately crash like before if I slice it after scaling the file further (under "Scale to size > uniformly...") to a size of 157mm or below (I tested a range of sizes from 40-200 mm, it consistently crashes at all sizes under 157mm).

enricoturri1966, 2018-04-13:
This issue should be fixed by: https://github.com/prusa3d/Slic3r/commit/359e7e4d32225d485c36c4733ba9f3847d5c80b5

bubnikv, 2018-04-13:
The fix will be part of Slic3r PE 1.40. Closing.

closed 2018-04-13 by bubnikv