prusa3d/PrusaSlicer #836: Feature request: Select the regions where support material rule should be applicable

ntsarb, 2018-04-12:
Hi,

This is a feature request. In some cases, support material is required only for particular features of an object. For example, in a particular design, I don't want to use support material below a particular layer, while it is needed from that layer up to a certain other layer.

It would be very useful to be able to select the regions of an object, wherein support material should be used (according to the relevant support material settings, angle threshold, etc).

Thank you

### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

supermerill, 2018-04-12:
If you use a modifier mesh
http://slic3r.org/blog/modifier-meshes
, specify an impossible overhang threshold (so no support for the object)
and set a good overhang threshold for the modifier mesh
Will it work?

bubnikv, 2018-04-12:
Sorry, this will not work.

I plan to work on supports in the 1.41 version, we hope to implement some support editing.

cerdeira, 2018-04-13:
That would be really great!

bubnikv, 2018-05-10:
I am closing this issue, I am certain there is a duplicate for it somewhere.

closed 2018-05-10 by bubnikv

ntsarb, 2018-05-10:
Somewhere? Can you please provide the relevant apen ticket number?

Thank you